openshwprojects/OpenBK7231T_App #812: Feature request - willing to help, just need direction

nospam2k, 2023-04-26:
I made an add in for Tasmota to act like a candle. I'd like to do the same for this project (if it doesn't already exist) I had to dig through the Tasmota code to find where to put it so it was driver independent.

Basic idea is I just set a range of random choice between red and green to make random yellows and random brightnesses at random speeds all between limiting parameters that could be set with Tasmota scene options. Where should I start looking?

openshwprojects, 2023-04-27:
Hello, so do that:
1. open drv_main.c , add new entry with name, with QuickTick, Init functions
2. create drv_candle.c , add QuickTick, Init functions
3. in QuickTick, use g_deltaTimeMS as time, include https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/quicktick.h
4. there, poll LED_GetEnableAll https://github.com/openshwprojects/OpenBK7231T_App/blob/main/src/cmnds/cmd_public.h 
5. then, do your calculations and, for example, use one of those functions to override color:
[image]

nospam2k, 2023-04-27:
Thanks, I will look into this more once I get my head around the project as a whole.

openshwprojects, 2023-04-29:
Let me know if you have any other questions. I can write a driver stub for you and let you continue...

nospam2k, 2023-04-29:
Thanks. Hopefully next week, I'm going to try to compile.

openshwprojects, 2023-04-29:
You can compile online in github.