openshwprojects/OpenBK7231T_App #814: Getting back wrong state of led_enableAll/get

nospam2k, 2023-04-28:
I'm unable to figure out how to request led_enableAll/get in order to get the correct response from the device.
Ex. The light is OFF and I send from mosquitto_pub:
```
openbk/lights/desk-light1/led_enableAll/get (null)
```
I get back:
```
openbk/lights/desk-light1/0/get 1
```

**Firmware:**
- Version 1.17.54
- Device?  Daybetter w/BP5758D_DAT/BP5758D_CLK
- Chip/model: BK7231T

openshwprojects, 2023-04-28:
Thank you for reporting issuei. It should be fixed now,c heck latest build. An appropriate self test has been also added to make sure that problem does not appear again in the future.

openshwprojects, 2023-04-29:
Hello, is it working for you now?

nospam2k, 2023-04-29:
Sorry for not getting back. I did test it and it's working. I switched over to using tele, but yes led_enableAll is working.

closed 2023-04-29 by nospam2k