openshwprojects/OpenBK7231T_App #815: Feature request - Art-net support

thirstyice, 2023-04-28:
Tasmota supports art-net, but this project doesn't seem to. Art-net would be super useful for using smart bulbs in installation art, for example

Wolfieeewolf, 2023-05-02:
Would be nice to also add E1.31 support as it's pretty standard for most light-controlling software.

There is the option to use DDP which depending on what program you're trying to use might work for you. 

It's not the fastest at updating but works well enough. Having other lighting protocols would most help speed things up.

DDP works with WLED, Xlights, Vixen Lights, and Hyperion.

Adding Art-NET and E1.31 would open this up to programs like OpenRGB and Artemis. Plus it would fix some issues with the slow updating of the LED's in Hyperion and WLED.

Anyway to add DDP just add  

startdriver DDP 

to the autoexec.bat and you should be able to get it working

thirstyice, 2023-06-04:
I'd be using it in live theatre, so art-net or sACN are basically my options. I only suggested art-net because tasmota has support for it, and I know there's a library for it, so hopefully adding it to this project wouldn't be as hard as building support from scratch.