openshwprojects/OpenBK7231T_App #817: Is 'tele/' necessary and what about 'cmnd' for Tasmota compatibility?

nospam2k, 2023-04-28:
I'm starting to find my way around OpenBK and I am SO happy for this project! I have a ton of Tuya bulbs that i've suffered with forever and am glad to be free!

I have a question about Tasmota compatibility. I've cobbled together my own node.js interface between Alexa and devices and am using Tasmota on several devices. I'm trying to understand some things in regard to OpenBK compatibility.

There is no need to add 'tele/' to the start of the MQTT topic in Tasmota. My testing shows that Tasmota doesn't acknowledge pub messages beginning with 'tele'. It only sends them. Is 'tele/' in the start of a topic necessary (as in can it be removed in the source) for messages published for OpenBK devices?

Also, Tasmota uses 'cmnd' before the various device functions, ie. 'cmnd/Color' 'cmnd/POWER'... shouldn't it be the same for OpenBK?

I realize I may be missing something here but just want to make sure.

openshwprojects, 2023-04-28:
Hello, can you be more precise, what do you mean with tele?

OBK from the very beginning accepts cmnd/obkDevName/commandText publishes, just like Tasmota.

So you can do cmnd/obkDevice/Power TOGGLE and it will work

Tele is published by OBK (if you enable it).

nospam2k, 2023-04-28:
Sure! I'll just use examples.

From mosquitto to a Tasmota device:

```
mosquitto_pub -h localhost -t tasmota/lights/davids-light/cmnd/STATE -m ""
```

Result from Tasmota device:
```
tasmota/lights/davids-light/STATE {"Time":"2023-04-28T20:20:25","Uptime":"43T23:01:50","UptimeSec":3798110,"Vcc":3.209,"Heap":32,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":43,"POWER":"OFF","Dimmer":100,"Color":"FFFFFF00","HSBColor":"180,0,100","White":0,"Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":4,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dafhome3","BSSId":"02:0C:43:26:46:A0","Channel":1,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":21,"Downtime":"0T02:00:37"}}
```

Notice there is no 'tele' preceding the topic and uses cmnd/STATE to return.

nospam2k, 2023-04-28:
Here is my Tasmota setup
[image]

nospam2k, 2023-04-28:
Ok, I see what is happening. I'm using 'tasmota/lights' as a PREFIX so I can sort through devices by prefix. I hadn't realized that %prefix% was important so I changed my Full Topic to %prefix%/tasmota/lights/%topic%/.

My misunderstanding of Tasmota!

closed 2023-04-28 by nospam2k