openshwprojects/OpenBK7231T_App #818: Bl602 Not saving config params

nospam2k, 2023-04-29:
I flashed a BL602L20 chip from a daybetter bulb following the tutorial using the lates Boufallo Dev Cube software. It comes up an AP and I can connect, but it won't save any config parameters.

[image]

openshwprojects, 2023-04-29:
Hello, does it also happen with older builds? Something from about a week ago?

nospam2k, 2023-04-29:
Can I downgrade OTA or do I need to reflash. I've removed all the Uart flash wires.

nospam2k, 2023-04-30:
Ok, I think it was just a corrupt flash. I installed a previous version, but I got better and shorter wires and that flashed ok. I then reflashed the latest version and it was ok. I'm going to close this.

closed 2023-04-30 by nospam2k