openshwprojects/OpenBK7231T_App #821: Energy reporting in kWh

rajil, 2023-05-05:
Hello,
Both Tasmota and Shelly device report energy in kWh. Can we get Openbeken to also report in kWh instead of Wh.

Thanks

openshwprojects, 2023-05-05:
Good idea, I can add a flag for that. I can't change it globally (but I could make this flag on by default for new builds) because many users already have it configured in Wh.

rajil, 2023-05-05:
Yes, that would work.

openshwprojects, 2023-05-05:
Ok, I am working on, will you be able to check if it works in few hours?

openshwprojects, 2023-05-05:
The new kWh flag, it should be added in latest build already.

rajil, 2023-05-05:
Thanks, it worked

rajil, 2023-05-06:
mqtt is still reporting in Wh instead of kWh.

rajil, 2023-05-06:
The sensor are still displaying in Wh in Home Assistant

[image]

This might be the reason  i am still not able to combine energy of Tasmota and Openbeken devices.

openshwprojects, 2023-05-06:
Have you done again the HASS Discovery?

rajil, 2023-05-07:
Thanks, after HASS discovery it is showing in kWh.