openshwprojects/OpenBK7231T_App #826: GreenSun Smart Plug not creating Wifi AP

TheNitek, 2023-05-14:
**Describe the bug**
I got a set of new GreenSun smart plugs/meters (see picture bellow), having a CB2S as a controller. Since they are not tuya devices, but VeSync, I had to use UART flashing, which worked fine. Unfortunately the device doesn't open a Wifi hotspot. I tried 1.17.104 and 1.17.113 software version, but it did not change. I also tried "Restore RF part" and "Erase all" before flashing, but the AP never appeared. Any ideas on how to get this device to work?

**Firmware:**
- Version: 1.17.104 and 1.17.113
- Device: GreenSun Smart Plug
- Chip/model: BK7231N
- Device config: None, didn't have the chance to configure anything

**Additional context**
[image]

openshwprojects, 2023-05-14:
What does the TX2 log say at 115200 baud?

gui flasher beta:
https://github.com/openshwprojects/BK7231GUIFlashTool/releases
can set SSID/PASS by the UART access, I wonder what would happen if you set your SSID there? Would it also fail to connect to AP, or is only the AP mode (not client) broken?

TheNitek, 2023-05-14:
I can't get anything out of TX2.

I tried the beta, couldn't read the config, but write it (and then also read it from then on). Still does not work.
On thing I noticed: Once powered the LED is always on, but at low brightness and after some time it switches to full brightness

openshwprojects, 2023-05-15:
ARe you really sure about that TX2 (or 2TX, whatever they call it)? 

The device, even with broken app code, just only with bootloader, should print chip name and some register on TX2 with each reboot.

openshwprojects, 2023-05-16:
This looks totally different than all the sockets I've seen. It might even have a different bootloader. It's not Tuya. Can you share 2MB backup?

TheNitek, 2023-05-16:
As mentioned it's a VeSync device, not a Tuya one, so I guess a different bootloader is quite possible
[readResult_BK7231N_QIO_2023-14-5--07-51-51.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/11490062/readResult_BK7231N_QIO_2023-14-5--07-51-51.zip)

divadiow, 2025-02-22:
and now https://www.elektroda.com/rtvforum/topic4107851.html