openshwprojects/OpenBK7231T_App #828: [Contact sensor] Deep sleep during initial configuration, after flashing the firmware

jacyker, 2023-05-14:
**Describe the bug**

The soldering and flashing of alternative firmware (Openbk7231n) were supposed to be the harder part. However, the problem is with the configuration because the device goes into what seems to be deep sleep mode after a few seconds, thus turning off the WiFi. I don't know who came up with this, but it's pure sadism because to configure the device, you have to keep "opening and closing doors" all the time. Is there anything that can be done about it? PowerSave 0 and DeepSleep 0 commands don't help.

**Firmware:**
- OpenBK7231N_QIO_1.17.113
- Tuya Contact Sensor
- Chip/model: CB3S/BK7231N

openshwprojects, 2023-05-15:
Hello, well, first of all, are you really sure it's OBK sleeping? Maybe it's TuyaMCu device and TuyaMCU is turning off the BK chip. Please post some photos.

jacyker, 2023-05-15:
You're right. It is most likely the case that CB3S is active only for a while (TuyaMCU protocol). Is it possible to get around this somehow programmatically, or by plugging the power supply directly into CB3S to set OpenBK7231T?

closed 2023-05-15 by jacyker

cocodrilabs, 2026-10-10:
For me the solution for the problem was to simply replace "Batt_Relay" with "AlwaysHigh". For this specific TUYA BK7231N it was Pin 14.

[image]