openshwprojects/OpenBK7231T_App #829: Devices lost connection with Hassio after Hassio restart.

FireFox1121, 2023-05-16:
**Describe the bug**
When I reboot server with hassio (scheduled daily reboot at night in cron), all connected BK7231/BL602 devices has lost hassio discover and in hassio this devices are "unknown".
And I need to click "discover" again in Hassio from device webUI configuration.

**Firmware:**
- Version 1.15.554
- Smart Power Switch
- Chip/model: BL602

**To Reproduce**
Steps to reproduce the behavior:
1. Connect device to hassio
2. Reboot hassio/server with hassio

**Screenshots**
[image]

**Additional context**
Scheduled Hassio re-discover..? (Hourly, Daily)

FireFox1121, 2023-05-16:
> check **Flag 2** in **configure General/Flags**
> 
> `Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)`

Thank you. Problem resolved.

closed 2023-05-16 by FireFox1121

openshwprojects, 2023-05-17:
> ID: <openshwprojects/OpenBK7231T_App/issue/829/issue_event/9257300324@
> github.com>

maybe also tick broadcast on connet flag?

FireFox1121, 2023-05-18:
> […](#)

Flag 10?