openshwprojects/OpenBK7231T_App #831: won't find/connect to my networtk

Ghome79, 2023-05-18:
device never connects to my network after i change to Wi-Fi instead of AP. even when i press scan in the config, my network doesn't show up unless i press scan over a dozen times and yet again go away

openshwprojects, 2023-05-18:
Hello, have you tried restore RF function from either flasher or the web app?

Ghome79, 2023-05-18:
changed MAC. still the same

Ghome79, 2023-05-18:
weird , when i used [OpenBK7231N_1.17.113 it worked fine. then updated it to 115. that when it all started and when i reverted to back 113 still had that problem ,i also tried rf restore.

openshwprojects, 2023-05-18:
Sorry but 1.17.115, from 2 days ago, and 1.17.113, from 5 days ago, has not many changes between them.
https://github.com/openshwprojects/OpenBK7231T_App/commits/main
nothing at all related to wifi/config has changed.

Maybe it's a fluke, issue happened at random, some outside condition affected firmware behaviour?