openshwprojects/OpenBK7231T_App #834: OTA from  1.14.104

StefanRied, 2023-05-23:
**Describe the bug**
Hi, I have a few devices flushed via UART cable to 1.14.104 and deployed them in my house.
Now, I realize that the OTA update seem not to work with this release. Could this be?
Maybe it is related to the strange blank in the MAC address?
Is there any trick to get the 1.14.104 accepting an OTA update to the latest version. I don't like unconnect them again and ...

**Firmware:**
- Version 1.14.104
- Device Sonoff Mini 3
- Chip/model: BL602
- Device config OTA Update.

**To Reproduce**
Steps to reproduce the behavior:
both the OTA in the App and the basic OTA URL seem not to work. The App says OTA Starting and nothing happens.

StefanRied, 2023-05-23:
Nothing really happens after this
[image]
waited after selecting the file. ... Then pressed START 
and the device seems tobe stuck.

openshwprojects, 2023-05-23:
It is possible that we've hit again the BL602 OTA size limit. It already happened once. I can try to fix it soon, I will disable extra HA discovery features on BL602. BL602 has less memory than BK7231 platforms and I never seen BL602+TuyaMCU, so removing extra TuyaMCU features won't hurt.

StefanRied, 2023-05-23:
much appreciated. 
Yeah, it feels like such a problem.

StefanRied, 2023-05-23:
it would also explain, that flashing via UART worked, while flashing OTA does not.

StefanRied, 2023-05-23:
I see you made the .127 which is smaller.
Still the same issue.

StefanRied, 2023-05-24:
I have very little knowledge about the BL60x platform. But hitting size/RAM limits in OTA is a problem I remember from other  devices such as the ESP. 
Maybe, we/somebody who understands the BL60x create a minimal OTA firmware, which is always small enough to be OTA-flashed over large firmwares that came via UART on the chip. This minimal OTA firmware can then accept a new release of the large firmware. 
It actually makes the OTA process more complicated in two steps. However, this would provide finally more RAM to the OTA update-able device. And it is definitely more convenient than dragging devices out of the wall to UART connect them again.

See as a sketch:
https://techoverflow.net/2021/11/09/minimal-platformio-esp8266-arduinoota-example/

Thanks for @openshwprojects or anybody else with time and knowledge ...

openshwprojects, 2023-05-24:
@StefanRied to the best of my knowledge, the latest build should work, maybe someone else can confirm/recheck. My BL602 dev board is currently out of commission because I've bricked it by trying to add MAC change feature. I must find my BL602 LED strip in parts bin and try myself. Maybe I will buy next BL602 dev board with donations if there is enough interest.

But in general, once something is working good for you, there is no need to OTA, we have a basic features set established and it works good, so there are not many improvements to be done on BL602...

Last weeks my focus was mostly on BK7231 flash tool and YT channel, so there is a lot to do and I must choose the priorities.
https://www.youtube.com/playlist?list=PLzbXEc2ebpH0CZDbczAXT94BuSGrd_GoM
https://github.com/openshwprojects/BK7231GUIFlashTool

StefanRied, 2023-05-24:
I have two BL 602 devices in my house of the .104 release. Both look for the DHCP like one. I guess this is the MAC blank problem. The router sends them the same IP and everything is messed up. I was hoping that I could avoid to to uninstall them and UART flash again. But anyway, thanks for your help. If I am the only one having problems, I'll go the UART way again and will be happy with one working version for many years....

openshwprojects, 2023-05-24:
Ota file went from 424 KB to 392KB... it should be enough, but maybe there is something else at play.

Related issue:
https://github.com/openshwprojects/OpenBK7231T_App/issues/745
Note the file size on the user's screenshot...

StefanRied, 2023-05-24:
https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.17.127/OpenBL602_1.17.127_OTA.bin.xz.ota 

is 406kB
401.892 bytes.

are we on the same file?

openshwprojects, 2023-05-24:
I might have looked at the zipped size on github page, not the unpacked one.

Please wait, I can try to lower the file size more. I can also try to find my BL602 magic strip and test myself.

StefanRied, 2023-05-24:
I think, I found one possible cause of the error. There is another partition table in the BouffaloLabDevCube-v1.8.4
It is named "partition_cfg_1M_ota.toml"

[image]

Please, don't waste time on this. 
I'll figure out if this was the cause and let everybody know, that other people do not fall into the same trap.

openshwprojects, 2023-05-24:
But... i looked there:
https://www.elektroda.com/rtvforum/topic3889041.html#gallery-6
And it seems I was using 2M version.

StefanRied, 2023-05-24:
Partition and Bootloader different. And the older DevCube version.
This explains a lot.
Looks like it's all my mistake or lack of knowledge. The BL chips are still new to me.
Thanks a lot for your insight and great help.