openshwprojects/OpenBK7231T_App #835: W600 cannot save settings

henkiejan1, 2023-05-24:
**Describe the bug**
I flashed before my W600 module and works fine. But with the new release i can put in wifi settings for x times but it looks like it saves nothing and booted just as AP... If i press on the scan wifi network button i got: W800 TODO. The downloaded latest version i got did not have the alpha extension...

**Firmware:**
- Version [OpenW600_1.17.127.fls]
- Device: Wavgat WIS600-01s
- Chip/model: W600
- Device config?

**To Reproduce**
Flash a W600 device and try to connect to wifi

openshwprojects, 2023-05-24:
Hello, the WiFi network scan is TODO, as described, but still, device should save settings.
I can try to find my W600 board and check.
In a meantime, can you flash by UART?
Can you try version from before April 20, 2023 and check if there also config save is broken?

henkiejan1, 2023-05-24:
Update: flashed the version op 19 april and that works! But needed to set static ip settings but ok. :)

openshwprojects, 2023-05-25:
So you're saying that newer version is broken? Are you able to tell when exactly it broke?

dy008, 2023-05-26:
I also have the same problem. After flashing into the 1.17.129 version, the wifi settings cannot be saved, only the AP function is enabled, and no configuration can be done. Flashing into 1.17.06 is normal

openshwprojects, 2023-06-14:
Fixed in:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20616695#20616695

closed 2023-06-14 by openshwprojects