openshwprojects/OpenBK7231T_App #844: Client to Tasmota 32 Matter Border Router

deepcoder, 2023-06-03:
**Describe the bug**
Wondering if you have looked at making OpenBK devices compatible with the Tasmota32 Matter Border Router system?
I tried attaching a OpenBK7231N device configured as just a simple Relay to a Tasmota32 12.5.0.4 device. The Tasmota32 device does not to seem 'handshake' with the OpenBK device. I have a Tasmota esp device at ip address 192.168.2.223 and the OpenBK device is at 192.168.2.220. The ESP32 Matter bridge seems to send out a 'status' request API to each, the Tasmota ESP device response, however the OpenBK device never seems to:

```
23:04:12.169 MTR: HTTP async-resp in 1112 ms from 192.168.2.223: [314] '{"StatusSTS":{"Time":"2023-06-...'
23:04:14.280 WIF: Checking connection...
23:04:14.291 MTR: HTTP async request 'http://192.168.2.223:80/cm?cmnd=Status+11'
23:04:14.297 WIF: DNS resolved '192.168.2.223' (192.168.2.223) in 0 ms
23:04:14.314 MTR: HTTP async request 'http://192.168.2.220:80/cm?cmnd=Status+11'
23:04:14.319 WIF: DNS resolved '192.168.2.220' (192.168.2.220) in 0 ms

**Firmware:**
- Version 1.17.132
- Device? 
- Chip/model: BK7231N
- Device config?
Thanks!

![matter01](https://github.com/openshwprojects/OpenBK7231T_App/assets/1409547/c70f4ae1-393c-49d3-ae0e-acf3f69bb50b)

openshwprojects, 2023-06-04:
That's a great idea, I can get it working for you, but can you provide more information?
Are you saying it sends: http://[192.168.0.200/cm?cmnd=Status+11](http://192.168.0.200/cm?cmnd=Status+11) ? 

Well, if so, my OBK seems to respond to that with full STATUS:
[image]

What is the expected response here?

How the full protocol looks like?

deepcoder, 2023-06-04:
Yes, I think the interpretation of what is to be returned by the 'Status 11' request is different. The Tasmota documentation (cited at bottom) is not really clear IMHO. The different in the returned JSON between the Tasmota and OpenBeken device is shown below. 

Is there anyway you could adapt the return to the 'Status 11' request to be same as Tasmota without breaking things?

Thank you again for your work on this project!

Tasmota 12.5.0
```
$ curl -s http://192.168.2.223:80/cm?cmnd=Status+11 | jq .
{
  "StatusSTS": {
    "Time": "2023-06-04T18:27:46",
    "Uptime": "2T20:31:40",
    "UptimeSec": 246700,
    "Heap": 18,
    "SleepMode": "Dynamic",
    "Sleep": 50,
    "LoadAvg": 19,
    "MqttCount": 1,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "gfi",
      "BSSId": "10:0C:6B:F6:30:E7",
      "Channel": 10,
      "Mode": "11n",
      "RSSI": 100,
      "Signal": -42,
      "LinkCount": 1,
      "Downtime": "0T00:00:03"
    }
  }
}
```

OpenBK7231N  1.17.132
```
$ curl -s http://192.168.2.220:80/cm?cmnd=Status+11 | jq .
{
  "Status": {
    "Module": 0,
    "DeviceName": "ew010-00",
    "FriendlyName": [
      "ew010-00_1"
    ],
    "Topic": "ew0101/00",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 1,
    "LedMask": "FFFF",
    "SaveData": 1,
    "SaveState": 1,
    "SwitchTopic": "0",
    "SwitchMode": [
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
    ],
    "ButtonRetain": 0,
    "SwitchRetain": 0,
    "SensorRetain": 0,
    "PowerRetain": 0,
    "InfoRetain": 0,
    "StateRetain": 0
  },
  "StatusPRM": {
    "Baudrate": 115200,
    "SerialConfig": "8N1",
    "GroupTopic": "",
    "OtaUrl": "https://github.com/openshwprojects/OpenBK7231T_App/releases/latest",
    "RestartReason": "HardwareWatchdog",
    "Uptime": 70442,
    "StartupUTC": "1969-12-31T04:25:58",
    "Sleep": 50,
    "CfgHolder": 4617,
    "BootCount": 22,
    "BCResetTime": "2022-01-27T16:10:56",
    "SaveCount": 1235,
    "SaveAddress": "F9000"
  },
  "StatusFWR": {
    "Version": "OpenBK7231N_1.17.132",
    "BuildDateTime": "Jun  3 2023 17:52:25",
    "Boot": 7,
    "Core": "0.0",
    "SDK": "obk",
    "CpuFrequency": 80,
    "Hardware": "BK7231N",
    "CR": "465/699"
  },
  "StatusLOG": {
    "SerialLog": 2,
    "WebLog": 2,
    "MqttLog": 0,
    "SysLog": 0,
    "LogHost": "",
    "LogPort": 514,
    "SSId1": "gfi",
    "SSId2": "",
    "TelePeriod": 300,
    "Resolution": "558180C0",
    "SetOption": [
      "000A8009",
      "2805C80001000600003C5A0A000000000000",
      "00000280",
      "00006008",
      "00004000"
    ]
  },
  "StatusMEM": {
    "ProgramSize": 616,
    "Free": 384,
    "Heap": 25,
    "ProgramFlashSize": 1024,
    "FlashSize": 2048,
    "FlashChipId": "1540A1",
    "FlashFrequency": 40,
    "FlashMode": 3,
    "Features": [
      "00000809",
      "8FDAC787",
      "04368001",
      "000000CF",
      "010013C0",
      "C000F981",
      "00004004",
      "00001000",
      "00000020"
    ],
    "Drivers": "1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45",
    "Sensors": "1,2,3,4,5,6"
  },
  "StatusNET": {
    "Hostname": "ew010-00",
    "IPAddress": "192.168.2.220",
    "Gateway": "192.168.0.1",
    "Subnetmask": "255.255.255.0",
    "DNSServer1": "192.168.0.1",
    "DNSServer2": "0.0.0.0",
    "Mac": "50:8a:06:b6:bf:e8",
    "Webserver": 2,
    "HTTP_API": 1,
    "WifiConfig": 4,
    "WifiPower": 17
  },
  "StatusMQT": {
    "MqttHost": "192.168.2.242",
    "MqttPort": 1883,
    "MqttClientMask": "core",
    "MqttClient": "ew0101/00",
    "MqttUser": "homeassistant",
    "MqttCount": 23,
    "MAX_PACKET_SIZE": 1200,
    "KEEPALIVE": 30,
    "SOCKET_TIMEOUT": 4
  },
  "StatusTIM": {
    "UTC": "1970-01-01T00:00:00",
    "Local": "1970-01-01T00:00:00",
    "StartDST": "2022-03-27T02:00:00",
    "EndDST": "2022-10-30T03:00:00",
    "Timezone": "+01:00",
    "Sunrise": "07:50",
    "Sunset": "17:17"
  },
  "StatusSNS": {
    "Time": "1970-01-01T00:00:00"
  },
  "StatusSTS": {
    "Time": "1970-01-01T00:00:00",
    "Uptime": "0T19:34:02",
    "UptimeSec": 70442,
    "Heap": 25,
    "SleepMode": "Dynamic",
    "Sleep": 10,
    "LoadAvg": 99,
    "MqttCount": 23,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "gfi",
      "BSSId": "30:B5:C2:5D:70:72",
      "Channel": 11,
      "Mode": "11n",
      "RSSI": 114,
      "Signal": -43,
      "LinkCount": 21,
      "Downtime": "0T06:13:34"
    }
  }
}
```
https://tasmota.github.io/docs/Commands/#status

```
Status
= show abbreviated status information
0 = show all status information (1 - 11)
1 = show device parameters information
2 = show firmware information
3 = show logging and telemetry information
4 = show memory information
5 = show network information
6 = show MQTT information
7 = show time information
8 = show connected sensor information (retained for backwards compatibility)
9 = show power thresholds (only on modules with power monitoring)
10 = show connected sensor information (replaces 'Status 8')
11 = show information equal to TelePeriod state message
12 = in case of crash to dump the call stack saved in RT memory

```
https://tasmota.github.io/docs/Commands/#teleperiod

```
TelePeriod
See current value and force publish STATE and SENSOR message
0 = disable telemetry messages
1 = reset telemetry period to firmware default (TELE_PERIOD)
10..3600 = set telemetry period in seconds (default = 300)
```

openshwprojects, 2023-06-05:
It seems that simply status 11 was not handled and used default case. I pushed a fix, but haven't checked with Tasmota. Can you check new build?

deepcoder, 2023-06-05:
Thank you very much for the updated firmware, I applied it. Unfortunately the Tasmota32 device and the OpenBeken device are still not handshaking on the Status=11 API request.

Unfortunately I am not currently setup to do a WireShark capture on this WiFi network, so I apologize for not being able to share very complete data.

First, below is a comparison of using curl to do the API request to both the OpenBeken device (ip address 192.168.2.220, version 1.17.133) and a Tasmota 8266 device (ip address 192.168.2.223, Tasmota 12.5.0). The JSON seems to be same format between the two devices. See comparison image below.

However, on a different side note, it appears that you are reporting the WiFi channel number as one higher than it should be. Both devices are connected to Channel 10 of the same WiFi router, however the OpenBeken device is reporting Channel 11.

Back to the core issue.

I have attached the console debug logs from both the Tasmota32 device and the OpenBeken devices. I now see the OpenBeken appearing to report :

```
Debug:CMD:cmd [Status 11]
Error:CMD:cmd Status NOT found (args 11)
```

in response to the Tasmota32 devices API request :

```
12:21:53.125 MTR: HTTP async request 'http://192.168.2.220:80/cm?cmnd=Status+11'
12:21:53.130 WIF: DNS resolved '192.168.2.220' (192.168.2.220) in 0 ms
12:21:53.146 MTR: HTTP async request 'http://192.168.2.223:80/cm?cmnd=Status+11'
12:21:53.151 WIF: DNS resolved '192.168.2.223' (192.168.2.223) in 1 ms
12:21:53.368 BRY: GC from 25015 to 15286 bytes, objects freed 103/247 (in 2 ms) - slots from 468/485 to 265/439
12:21:53.451 MTR: HTTP async-resp in 299 ms from 192.168.2.223: [314] '{"StatusSTS":{"Time":"2023-06-...'
```

Note the different logs of the response that the Tasmota32 device shows from the Tasmota 8266 device (ip 192.168.2.223) and the OpenBeken device (ip 192.168.2.220)

Other than WireShark, is there a way I can display what the OpenBeken devices is getting as the API request?

Thanks again for your help.

```
$ curl -s http://192.168.2.220:80/cm?cmnd=Status+11 | jq .
{
  "StatusSTS": {
    "Time": "2023-06-05T19:25:53",
    "Uptime": "0T01:33:58",
    "UptimeSec": 5638,
    "Heap": 25,
    "SleepMode": "Dynamic",
    "Sleep": 10,
    "LoadAvg": 99,
    "MqttCount": 23,
    "POWER": "ON",
    "Wifi": {
      "AP": 1,
      "SSId": "gfi",
      "BSSId": "30:B5:C2:5D:70:72",
      "Channel": 11,
      "Mode": "11n",
      "RSSI": 110,
      "Signal": -45,
      "LinkCount": 21,
      "Downtime": "0T06:13:34"
    }
  }
}
$ curl -s http://192.168.2.223:80/cm?cmnd=Status+11 | jq .
{
  "StatusSTS": {
    "Time": "2023-06-05T20:26:01",
    "Uptime": "3T22:29:55",
    "UptimeSec": 340195,
    "Heap": 18,
    "SleepMode": "Dynamic",
    "Sleep": 50,
    "LoadAvg": 19,
    "MqttCount": 7,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "gfi",
      "BSSId": "10:0C:6B:F6:30:E7",
      "Channel": 10,
      "Mode": "11n",
      "RSSI": 100,
      "Signal": -42,
      "LinkCount": 1,
      "Downtime": "0T00:00:03"
    }
  }
}

```
[image]

deepcoder, 2023-06-05:
As a follow up to my prior post. I am now seeing that 'Status' or 'Status=11' are not listed as commands in the documentation. They seem to yield results when call as a HTTP API, but generate the following in the log :

```
Debug:CMD:cmd [Status 11]
Error:CMD:cmd Status NOT found (args 11)
```
Additionally, neither the 'Status' or the 'Status=11' command works in the 'console', entering 'Status=11' yields:

```
Debug:API:POST to api/cmnd
Debug:CMD:cmd [Status=11]
Error:CMD:cmd Status=11 NOT found (args )
Info:CMD:[WebApp Cmd 'Status=11' Result] Unknown command
```
Do I not have the device configured correctly?

Thx!

openshwprojects, 2023-06-05:
I'll have more time to look into it tomorrow morning but:
- status handling is kinda tricky so it will cause a misleading console message, you should test it just with a web browser
- wifi channel 11 is a hardcoded placeholder which stays in place forever, we need to get correct channel from SDK
- if you want to display what obk receives, you can just do a fork and add a console print in HTTP handler, you can also compile obk in windows simulator mode and run it on windows with msvc and just use debugger

s-hadinger, 2023-06-05:
For Matter the wifi channel is not used

deepcoder, 2023-06-05:
I am now seeing that the Tasmotadmin utility, while seeming to report on the status of this OpenBeken device, also generate the following in the log of the OpenBeken device:

```
Debug:CMD:cmd [status 0]
Error:CMD:cmd status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)
Debug:CMD:cmd [Status 0]
Error:CMD:cmd Status NOT found (args 0)

```

On another side topic, I note that the OpenBeken device reports RSSI values in excess of 100, I believe that Tasmota defines this RSSI field as a percentage between 0 and 100 %.

```
  "StatusSTS": {
    "Time": "2023-06-05T21:42:31",
    "Uptime": "0T00:52:50",
    "UptimeSec": 3170,
    "Heap": 25,
    "SleepMode": "Dynamic",
    "Sleep": 10,
    "LoadAvg": 99,
    "MqttCount": 23,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "gfi",
      "BSSId": "30:B5:C2:5D:70:72",
      "Channel": 11,
      "Mode": "11n",
      "RSSI": 112,
      "Signal": -44,
      "LinkCount": 21,
      "Downtime": "0T06:13:34"
    }

```

deepcoder, 2023-06-05:
@s-hadinger can I send you a couple to play with gratis. Not that you don't have enough on your plate. I bought a case of Tuya compatible RGB strips for pennies on the dollar that easily converted to OpenBeken. My significant other gave me that 'look' when the box arrived...

s-hadinger, 2023-06-06:
Thank Dave. Let me find one I can easily get from Europe. Any suggestions on a device that can be easily flashed and is available from amazon.fr for example?

openshwprojects, 2023-06-06:
@s-hadinger you can run OpenBeken on Windows in Simulator mode: 
[image]
@s-hadinger here is how header is created: https://github.com/openshwprojects/OpenBK7231T_App/blob/6de16f079237b25aec3af5eb6467c623939407c6/src/httpserver/new_http.c#L219
Is chunked mode required? We do not currently send content-lenght because we had some kind of optimisation issue some time ago and we decided it's more optimal to send data on the fly to LWIP instead of making a large buffer to keep whole reply and then to process it

s-hadinger, 2023-06-06:
Thanks. No chunked mode is not required, although it may render the request faster since we don't need to wait for the TCP connection to close to detect the end of the response.

I fixed a bug in the Tasmota Matter, I hope it works now.

openshwprojects, 2023-06-07:
Very good! Does that mean the support is now useable in practical cases, or do we need to add anything more to OBK? Let me know, I can help if that's needed, @deepcoder .

Btw, if you have some free time, you can also help us by doing a short writeup how to do that _Apple HomePods turning the OpenBeken Relay_  connection and post it as a guide, we have a list programme for people contributing: https://www.elektroda.com/rtvforum/topic3950844.html

s-hadinger, 2023-06-07:
From the feedback and the logs, it looks to be working fine. I don't see any required change on OBK. And response time was great (~150ms) despite the lack of Content-Type or Chunked. This is excellent news.

s-hadinger, 2023-06-13:
This feature is now stabilized in Tasmota. Here is how the new UI looks like:
[image]

I suppose you can close the issue