openshwprojects/OpenBK7231T_App #845: Home Assistant auto discovery does not work

Heisenberg2980, 2023-06-04:
**Describe the bug**
I flashed couple of BK7231N switches and just realised the Home Assistant auto discovery does not work as Home Assistant doesn´t discover the switches after pressing "Start Home Assistant Discovery". I am not very confident with how auto discovery works, but looking at the MQTT messages I realised other devices send "homeassistant/switch/DEVICE_ID/switch/config" instead of "homeassistant/switch/DEVICE_ID/config" that these devices are sending, so the issue might be the missing "switch" after the device ID.

Another small issue I found is that the default topic (Client Topic (Base Topic)) is just the device ID so that creates a lot of noise in MQTT as every device has their own topic, instead it would be better to have a common base before device ID. Thankfully that can be change in the MQTT settings so I have change the base topic in my devices to "OpenBK7231/DEVICE_ID". I think this should be the base topic by default.

P.S.: Thank you very much for this amazing project, finally my Tuya devices are local!!

**Firmware:**
- Version 1.17.131
- Device? Light Switch King Smart Switch Tuya Panel 2021
- Chip/model: BK7231N
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Config, Home Assistant Configuration
2. Click on Start Home Assistant Discovery
3. Open Home Assistant and see that it has not discovered the device

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-06-12:
Hello, which HA version are you using? Are you sure that under default config, the discovery does not work? Maybe they had changed something in the standard, because I have tens of devices and it works every time for me.

Heisenberg2980, 2023-06-12:
You are right, I was looking at the wrong place

closed 2023-06-12 by Heisenberg2980