openshwprojects/OpenBK7231T_App #847: Home Assistant integration without the need of an MQTT broker

xaviergr, 2023-06-07:
MQTT isn't always desired for various reasons. It would be nice to have OpenBeken to be able to support direct communication with Home Assistant just like esphome devices do.

openshwprojects, 2023-06-08:
I think I saw some time ago a HA plugin that used Tasmota JSON API to communicate with device via HTTPP. Would that suit your need?

xaviergr, 2023-06-08:
That's a good point, initially I was using a rest command in HA that was called as a service, but I only found how to toggle the relay from the web interface of OpenBK (http://device_ip/index?tgl=0). With your comment I searched and found out that I can do something like "http://device_ip/cm?cmnd=power%20on" to specifically set the relay state, which is exactly what I was looking for.

For now this is all I need, but of course if you keep adding devices, manually setting all the needed rest commands in HA can be quite a hassle. I wonder if the plugin you mention can do auto-discovery and automatic configuration of mqtt devices.