openshwprojects/OpenBK7231T_App #849: Device Group bug

Jonsonmon, 2023-06-11:
Device Group between 2 OpenBeken devices is not working properly.
It's working about 3/5 of the times.
The other device just don't show any DGR logs sometimes.
I Have tasmota devices and it working flawlessly there,

**Firmware:**
- Version: 1.17.135
- Device: AVATOO 1 Gang Wall light switch
- Chip/model: BK7231N
-

openshwprojects, 2023-06-12:
Hello, does it work well some amount of time and then breaks, or is that 3/5 a random chance that each packet gets to the destination?

Jonsonmon, 2023-06-12:
Hi, thanks for your reply.
it's a 3/5 random chance that each packet gets to the destination.

Jonsonmon, 2023-06-12:
Found a solution as a workaround until the Device Group bug will be resolved...
First clean the Device Group fields. and then stop DGR Driver with the following command:
`stopDriver DGR`
Then, add the following lines to autoexec.bat:
```
addChangeHandler Channel1 == 0 SendGet http://<IP-OF-OTHER-SWITCH>/cm?cmnd=Power1%20Off
addChangeHandler Channel1 == 1 SendGet http://<IP-OF-OTHER-SWITCH>/cm?cmnd=Power1%20On
```

Done. works flawlessly now.

Wolfieeewolf, 2023-08-11:
Did we work this out??

I am having this issue as well. I have 6 downlights all BK7231N. I can't get any of them to talk to each other. Nothing appears in the logs.

Would be great to see something more like how WLED handles [Virtual LEDs](https://kno.wled.ge/advanced/ddp/) but you could make it for any device.  Something like having a master light that receives all the info then the rest of the group would be slaves. That way you can then group that master light into another group if needed.  

I have been trying to group all the downlights together then using a OpenOK switch trying to use a SendGet command to one of the downlights and having them all turn on and off. I have not been successful yet. I have tried just using the SendGet command to talk to all the lights but it appears SentGet only works with one light at a time. I tried using backlog but it just cycles the lights. 

I have been using WLED to control all the light as a group for now but the downside is that I can link them to my light switch and that DDR is someone slow and doesn't always work well.  Would be great to see OpenBK be something like a watered down version of WLED with how it handles RGB lights but I also understand that most BK chips are only 2mb and WLED need at least 4mb of space.  This would also mean having separate RGB builds of openBK so that devices that don't use RGB don't get filled up with stuff they don't need.

Anyway getting off track. 

Basically, it appears OBK to OBK device grouping is broken. I can't get it to work any of the time. There is nothing in the logs but on the main screen, it says it has sent the packets but nothing is begin received at the other end.

maxb80, 2023-08-22:
The same issue for me. I have two Gosund SW5 with OpenBK flashed on them. I configured Device Group with the name `hallway-light` and only one option enabled for `Power Receive` and `Send` on both switches. I attached the debug logs and as yo can find that the latest `DGR:Power event` has `OFF` message and `ON` is missing.
[gosund.log](https://github.com/openshwprojects/OpenBK7231T_App/files/12411071/gosund.log)

**Firmware:**
- Version: 1.17.191
- Device: GHome (Gosund SW5)1 Gang Wall light switch
- Chip/model: BK7231T