openshwprojects/OpenBK7231T_App #851: [Question] How do I change the NTP Server?

viny182, 2023-06-13:
Hello Guys,

In a BK7231N chip flashed with version 1.17.54, how do I setup a manual NTP Server?

I try to run the command NtpServer on the console, but I'm getting the command unknown error, as below...
[image]

Also, on the homepage, I See the NTP server enabled is the 217.147.223.78 from pool.ntp.org. This what I need to change.

[image]

ataa, 2023-06-13:
Try:
`ntp_setServer` command.

viny182, 2023-06-13:
> Try:
> 
> `ntp_setServer` command.

Unfortunately I've got the same output: 'Unknow command'

ataa, 2023-06-13:
I am on version 1.17.104 (BL602) and it works on mine, make sure ntp driver is loaded.

`ntp_setServer 192.168.20.1`
```
Info:NTP:NTP server set to 192.168.20.1
Info:CMD:[WebApp Cmd 'ntp_setServer 192.168.20.1' Result] OK
```

viny182, 2023-06-13:
> I am on version 1.17.104 (BL602) and it works on mine, make sure ntp driver is loaded.
> 
> 
> 
> `ntp_setServer 192.168.20.1`
> 
> ```
> 
> Info:NTP:NTP server set to 192.168.20.1
> 
> Info:CMD:[WebApp Cmd 'ntp_setServer 192.168.20.1' Result] OK
> 
> ```
> 
> `ntp_info`
> 
> 
> 
> ```
> 
> Info:NTP:Server=192.168.20.1, Time offset=0
> 
> Info:CMD:[WebApp Cmd 'ntp_info' Result] OK
> 
> Info:NTP:Seconds since Jan 1 1900 = 3895663524
> 
> Info:NTP:Unix time  : 1686674724
> 
> Info:NTP:Local Time : 2023/06/13 16:45:24
> 
> ```

Errmmm, my bad.

I tried in other device where NTP driver was stopped.

I confirm Starting NTP and runing ntp_setServer x.x.x.x works.

Thank you @ataa !

closed 2023-06-13 by viny182