openshwprojects/OpenBK7231T_App #852: BL602 with BL0937 IC no power/voltage metering

antoniodelavega, 2023-06-14:
**Describe the bug**
I have an Elivco plug with power meter.
I have installed the OpenBK7231T/OpenBeken release 1.17.136,
The plug has a small board with a BL602 and a BL0937 in a bigger board to get information about the powerr and the voltage.
I have checked the connection of the pins with a multimeter.
The LED, the button and the relay worked fine, but all the information about the power meter is at 0

I found others with the same issue:
https://www.elektroda.com/rtvforum/topic3945009.html#20357626

**Firmware:**
- Version 1.17.135
- Device Elivco Plug
- Chip/model: BL602

openshwprojects, 2023-06-15:
This is basically a feature request, as we don't have a counter in BL602 yet. See my reply on Elektroda.

openshwprojects, 2024-04-05:
It has been recently solved on Elektroda. BL602 supports BL0937 now.

closed 2024-04-05 by openshwprojects