openshwprojects/OpenBK7231T_App #855: Startup command separator needs space before and after - documentation unclear

sven337, 2023-06-17:
**Describe the bug**
I used the following startup command for my device:
`startDriver BL0942; PowerSave 1`
But this didn't seem to load the driver at startup.
The inline documentation suggests it should:
`Use backlog cmd1; cmd2; cmd3; etc to enter multiple commands`

After testing in "execute custom command", the problem is obvious:
`Info:MAIN:Driver BL0942; is not known in this build. `

What worked is to add a space before the semicolon. I assume this is a documentation error and we should just add a space so it reads:
`Use backlog cmd1 ; cmd2 ; cmd3 ; etc to enter multiple commands`
 ?

openshwprojects, 2023-06-17:
Hello, are you sure that the following startup command:
`backlog startDriver BL0942; PowerSave 1`
is not working?
I've added a self test and it seem to work:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a395ef2cdf9a2c418cc25ae4989392ba5fdbc131
Maybe you are just missing "backlog" command? Well.... maybe it should be on by default!

sven337, 2023-06-17:
I am not using "backlog". I did not know what it was for so I tried without :)
What does it do?

openshwprojects, 2023-06-17:
It splits string by ; and executes multiple commands.

sven337, 2023-06-17:
Oops. Seems obvious what I did wrong then, sorry for bothering you.
Closing

closed 2023-06-17 by sven337