openshwprojects/OpenBK7231T_App #856: PinDeepSleep doesnt seem to do anything

jmacato, 2023-06-17:
**Describe the bug**
I have a presence sensor but the DoorSensor driver doesnt seem to have a way to have its timer reset so i was trying to do a PinDeepSleep instead. However, it doesnt seem to really put the MCU to sleep at all. Here's my script if it could help:

```
startDriver TuyaMCU
startDriver tmSensor
waitFor MQTTState 1

alias publishPresence SetChannel 50 $CH0; publish presence $CH0
alias extendTime publish test1 extending; DSTime 100
alias gotoLoop goto loop
publishPresence

addChangeHandler Channel0 == 0 publishPresence
addChangeHandler Channel0 == 1 publishPresence

SetChannel 50 $CH0

loop:
     delay_s 1
     if $CH0==1 then extendTime
     if $CH50==$CH0 then gotoLoop
     publishPresence
     if $CH0==0 then PinDeepSleep
     goto loop
```

**Firmware:**
- Version 1.17.130
- Device - Custom
- Chip/model: BK7231N
- Device config: 

```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "0",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "7": "WifiLED;0",
    "8": "dInput;0"
  },
  "command": "",
}
```

openshwprojects, 2023-06-17:
Hello, have you tried on a more simple setup?

Just make sure you have one button (to wake up) and assign PinDeepSleep from console. .Button press should wake up the device.

jmacato, 2023-06-17:
hmm, manually using PinDeepSleep in console and it worked, perhaps there's an issue with scripts?

jmacato, 2023-06-17:
yeah, i think `if $CH0==1 then PinDeepSleep` doesnt seem to work

jmacato, 2023-06-17:
Yeah, does PinDeepSleep work on scripts at all? I feel like it isnt working there

openshwprojects, 2023-06-17:
It should work, just like all commands.

Why are you trying to use tmSensor with pindeepsleep? tmSensor is used on devices where TUyaMCU controls power of wifi module, so no deep sleep is needed.

jmacato, 2023-06-17:
@openshwprojects I have removed those startDriver calls, i thought they were essential for PinDeepSleep but i guess not.. 

Right now i just made it so that im just sending the PinDeepSleep over MQTT.. But yeah, no dice on making PinDeepSleep work on scripts here.

openshwprojects, 2023-06-18:
Can you post a photo of your device?

jmacato, 2023-06-18:
[image]

It formerly has a XB809 chip in it but i replaced it into a CB8P chip from a dead smart rgb bulb. I wired the PIR's output according to the config i posted in the issue

openshwprojects, 2023-06-18:
This is not a TuyaMCU device. So at least we know that.

What kind of device behaviour do you want?

I think you can use DoorSensor driver for that, you just need a proper DSEdge setting.

Make sure to also have a Button role on button pin so you can wake up device with button.

Btw DSTime can be also used to add time with a button. For example:

`addEventHandler OnClick 8 DSTime + 100`

this will, on click of button on p8, add 100 seconds of door sensor time before sleep... so this feature is already present.

jmacato, 2023-06-18:
well the behavior i wanted to implement is for the device to not sleep while the Sensor/Ch0 is high or 1, then when it goes to 0 it should wait for a minute before deep sleeping until Ch0 is high again

closed 2023-06-25 by jmacato