openshwprojects/OpenBK7231T_App #859: Missing Hue integration - smart bulbs can not be controlled from Alexa devices

MnM001, 2023-06-18:
Hi,

It seems that at this point in time only WEMO integration is enabled. Alexa finds my smart lights as switches. I can only turn them on or off from Alexa. I cant control color, dimming etc.

Will be good to have Hue integration working so Alexa can correctly identify smart lights as lights and be controlled accordingly by Alexa devices.

Any plans or ETA for Hue integration?

Cheers.

openshwprojects, 2023-06-18:
Yes, it's on TODO list, and btw, we can need some testers for that, so please be around for the time being and maybe you can help if you want.

MnM001, 2023-06-19:
Hi,

Sounds good that this is on the radar.
Happy to help with testing. Just let me know when.
Cheers.

CLAM01, 2023-07-24:
I just saw this feature request.
This is the same topic as #885.

I am also happy to test LSC Mood Light from Action Markt Netherlands.

https://www.action.com/de-de/p/lsc-smart-connect-stimmungslampe/

If this feature works, then I could control everything via Alexa in my home.

CLAM01, 2023-08-29:
Hi, is there already a roadmap or milestone planning for when the driver will be finalized?

openshwprojects, 2023-08-29:
I will try to look into it again soon, as said on forum. I am trying to solve issues one by one but I am just one person. Do you know C and would like to help?

CLAM01, 2023-09-20:
Hi, unfortunately I don't know C or any other programming language, otherwise I would love to support.

Milinator, 2024-10-15:
@openshwprojects I would like to help to add Hue Emulation Support. I would even pay money to make it work.