openshwprojects/OpenBK7231T_App #860: Use a configured ShortName for DHCP requests

tsybulin, 2023-06-19:
**Describe the bug**
Currently, DHCP requests use the auto-generated hostname (which does not carry a semantic load and makes it difficult to find the device in the list)

[image]

instead of the short name specified in the configuration

[image]

**Firmware:**
- Version [e.g. 1.17.135]
- Chip/model: BL602

openshwprojects, 2023-06-20:
Hello, it seems like a BL602 issue. For Beken, we have it implemented:
[image]
So basically we need to look into BL602 SDK and figure out how to change the host name.

ataa, 2024-02-01:
Found this in Ai Thinker WB2 (Bl602) SDK, can we use this to change the default hostname to device name?

[image]

https://github.com/Ai-Thinker-Open/Ai-Thinker-WB2/blob/c86f5aed43fd1b9bd1918e9bd76ed561a13c9120/components/network/wifi_manager/bl60x_wifi_driver/wifi.c#L205