openshwprojects/OpenBK7231T_App #861: CBU-NL Flashing Help Required

Nailik, 2023-06-20:
I have troubles backing up the firmware of the CBU-NL with the GUI tool.

According to https://developer.tuya.com/en/docs/iot/CBU-NL-module-datasheet?id=Kbaeq6j53y0yg 
it uses the BK7231N chip.

I traced the following pins (also according to data sheet).

I tried using reset via the CEN pin and also power on/off and also different Baud Rates but i always get the error:

getting bus failed

[image]

openshwprojects, 2023-06-20:
Hello, this looks like not a bug report, but like an end-user per-device help request, so it belongs to our forum:
https://www.elektroda.com/rtvforum/topic3984948.html
Let's continue discussion there.

Nailik, 2023-06-20:
fixed anyway, i used rx01 and tx01 the rx tx to programm are on the other side.

closed 2023-06-20 by Nailik