openshwprojects/OpenBK7231T_App #866: Feature Request: White Blend Mode for RGBW lights

neominik, 2023-06-27:
I have a MagicHome RGBW LED strip where the white LED is neither cold nor warm, but pretty neutral. The default firmware uses the red and blue LEDs in addition to the white channel to mix the proper temperature. Here, I can use flag 24, but that assumes the white to be warm white, which is not the case. I also can't use the normal RGBCW mode, because warm white is just off (unmapped channel) in that case.

Tasmota has a solution for this, namely the [white blend mode](https://tasmota.github.io/docs/Lights/#white-blend-mode), where the light is exposed as RGB, but those values get the white led mixed in to improve brightness and color output. It works pretty well, but unfortunately not on the BL602 :(
I'm not sure how easily such a mode can be replicated in this project, since I haven't really figured out the codebase yet, but this would be great to have. Most people seem to have RGBCW lights, but I've stumbled across comments from other people with similar problems in other issues as well. Unfortunately, none of those have a concrete solution.

openshwprojects, 2023-06-28:
Hello, so you are basically looking for modified flag 24? I haven't been aware about a "non cold non warm" RGBW lights, but maybe it could work with some modifications to flag 24 handling?

As far as I remember, we still have to improve the flag 24 itself, there have been some reports saying that colour fractions in C emulation should be improved.

neominik, 2023-06-28:
I'm not sure if we should modify flag 24 add a separate flag or modify 24 with a separate flag. I guess 24 might make sense for some people as is, but I don't know how (much) it is being used.
The behaviour of the proposed white blend mode (as implemented in Tasmota) is actually pretty simple:
Convert an incoming RGB value to RGBW in this way: `W = min(R, G, B); R = R - W; G = G - W; B = B - W`

panchomira, 2023-07-25:
Hello! Something similar happened to me. This is my documentation about the issue (https://github.com/openshwprojects/OpenBK7231T_App/issues/763)