openshwprojects/OpenBK7231T_App #868: OAT 802.1X supplicant took to long

gitisgreat2023, 2023-06-29:
Found access point name: "SmartLife-9CED", trying to connect...
Error: Connection activation failed: 802.1X supplicant took too long to authenticate.

I don't understand the error: is it coming from connecting to the unsecured plug AP? Or is it from trying to connect to my own WiFi?

**Firmware:**
- 1.1.8 (checked in app)
- Girier - JR-PM01
- Chip/model: should be Girier - JR-PM01 (didn't open it)

**To Reproduce**
Steps to reproduce the behavior:
1. ./tuya-cloudcutter.sh -v

openshwprojects, 2023-06-29:
Hello, do you have other OBK devices to check is it a device problem or a firmware problem?

gitisgreat2023, 2023-06-30:
> Edit: Are you sure this problem is OBK related? It looks like.... like you're trying to use cloudcutter.

sorry, yes, cloudcutter.... sorry, wrong forum, my posting can be removed!