openshwprojects/OpenBK7231T_App #869: Flashing from ESP

robertscheib, 2023-06-30:
I currently have a MI-BW210-999WW that I used Cloudcutter to flash ESP but I would like to try this firmware.

I attempted to use the OTA upload function on the Web portal for the light but all firmware options here generate a failure.

Am I able to move to OpenBekken?

openshwprojects, 2023-06-30:
Hello, can you elaborate on what you mean? Is your device Beken or ESP8266? I will temporarily close it, but I will try to help if you provide more info.

closed 2023-06-30 by openshwprojects

aidbish, 2023-07-01:
i think he means esphome

robertscheib, 2023-07-01:
Sorry for the lack of details. It currently has esphome installed which was installed when I used Cloudcutter.

I would like to install OpenBekken on it.

crisnar, 2023-07-26:
Hi, Also I'm looking for a way to install via openbeken ota on a cb3s chip smart plug where esphome is installed.  Unfortunately I can no longer disassemble the plug because I had to glue it, so I don't have the possibility to install via serial.  Is it possible that there is no way to do this?