openshwprojects/OpenBK7231T_App #870: cosumption_stats is too big for HA to use it, here is the error generated by HA

DeDaMrAzR, 2023-07-06:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version [e.g. 1.14.129]
- Device? 
- Chip/model: BK7231T?
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-07-06:
@valeklubomir any thoughts on this? Was it working for you with HA? I could lower the amount of digits after decimal point, but it still won't strip enough chars, or will it?

openshwprojects, 2023-07-07:
Wait, even if I trim totally samples, it's still over 300 chars. What has happened here? @valeklubomir  , were you using that with HA?

`{"uptime":2178232,"consumption_total":15203.40039,"consumption_last_hour":2.04851,"consumption_stat_index":35708,"consumption_sample_count":60,"consumption_sampling_period":60,"consumption_today":100.72854,"consumption_yesterday":295.76525,"consumption_clear_date":"2023-04-28T01:01+02:00","consumption_samples":`