openshwprojects/OpenBK7231T_App #872: Only responds to GETs and not POSTs

spacelama, 2023-07-10:
Particularly when configuring a device, it's normal to accept POST rather than just GET.  For instance, this works with tasmota:

```
> curl --data-urlencode "cmnd=Power1 Toggle" http://studio-power/cm 
{"Time":"2023-07-10T15:48:04","Uptime":"104T15:39:26","UptimeSec":9041966,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":77,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Asio2.4","BSSId":"B0:95:75:E8:CA:3D","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":20,"Downtime":"0T00:10:49"},"POWER1":"ON"}                                                                                                                                    
> curl --data-urlencode "cmnd=Power1 Toggle" http://studio-power/cm 
{"Time":"2023-07-10T15:48:06","Uptime":"104T15:39:28","UptimeSec":9041968,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":77,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Asio2.4","BSSId":"B0:95:75:E8:CA:3D","Channel":1,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":20,"Downtime":"0T00:10:49"},"POWER1":"OFF"}    
```

But with beken, that action is ignored and no status (or error!) is returned:
```
> curl --data-urlencode "cmnd=Power1 Toggle" http://four-power/cm 
```

We can force curl to send GETs rather than POST with `-G`, and it works:
```
> curl -G --data-urlencode "cmnd=Power1 Toggle" http://four-power/cm 
{"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF"}                                                                            > curl -G --data-urlencode "cmnd=Power1 Toggle" http://four-power/cm 
{"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF"}                                                                           ```
But not all software can be configured that way, so it would be better for openbeken to support all commands as POSTs as well.

spacelama, 2023-07-10:
I begun tracing the code, and without compiling this myself and trying it out, I'm suspicious of new_http.c:592:
```
        for (i = 0; i < sizeof(methodNames) / sizeof(*methodNames); i++) {
```

I would have expected sizeof(methodNames) to be 4, not the size of the bytes making up that const char* array.

openshwprojects, 2023-07-10:
Thank you for noticing that missing POST. I was not aware about it.

However, I don't think it's the sizeof fault.

sizeof(methodNames) is a byte size of whole array, let's say 8 for both post and get.
sizeof(*methodNames) is a byte size of first element, so 4.
8 / 4 is 2, we have two methods.

I think the issue may be somewhere else. For example, in cm handler, we are explicitly taking command string from GET argument. I should be modified to check for POST data as well.

PR #900 (2023-08-16): Respond to POSTs as well as GETs in /cm handler
Not sure my adding a new http_getRawArg method is the best way to go about ignoring the initial "?" when parsing the data within the body of a POST request, but otherwise this does seem to fix #872 for me.

Fixes #872
closed 2023-08-16 by openshwprojects