openshwprojects/OpenBK7231T_App #873: MQTT Discovery (Home Assistant) is missing required key

skellycode, 2023-07-10:
**Describe the bug**
Using the default Home Assistant Discovery option in the OpenBK config of a device produces an error in the HA log.  It appears that the 'State Topic' (a required field) is missing.  (Not sure if this is for power monitoring devices only?)

**Firmware:**
- Version - 1.17.165
- Device - PC191HA Smart Plug with Power Monitoring
- Chip/model - BK7231T

**Error Log**
```
2023-07-10 20:51:35.703 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'device': {'identifiers': ['DryerPlug_BF882CC8'], 'name': 'DryerPlug', 'sw_version': '1.17.165', 'manufacturer': 'Beken Corporation', 'model': 'BK7231T', 'configuration_url': 'http://10.50.0.81/index'}, 'name': 'http://10.50.0.81/index', 'availability_topic': 'DryerPlug/connected', 'unique_id': 'DryerPlug_BF882CC8_sensor_8', 'qos': 1, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 270, in async_discover
    config: DiscoveryInfoType = discovery_schema(discovery_payload)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 232, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 355, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.11/site-packages/voluptuous/validators.py", line 351, in _exec
    v = func(v)
        ^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: required key not provided @ data['state_topic']
```

openshwprojects, 2023-07-10:
Hello, which HA version? A lot of users have paired power metering devices with HA so far, so it looks like HA must have changed something?

Can you test with power metering disabled, does the problem persist?

skellycode, 2023-07-11:
HA Version 2023.6.3
Looks like it might be regarding the `sensor.DEVICENAME_consumption_stats` sensor specifically, as this appears as 'Unknown' in HA.

DeDaMrAzR, 2023-07-12:
@skellycode there is a hard limit for State in HA to 255 characters - consumption stats is affected by this.

skellycode, 2023-07-12:
@DeDaMrAzR Ah, yes that looks to be the issue - thanks for pointing that out 👍 
To avoid continued log spam, does this need to be accounted for on the OpenBK side of the fence, or the HA side?  Not sure what needs to be 'fixed' here 🤔

DeDaMrAzR, 2023-07-12:
@skellycode to be honest I am not sure, I would think 255 limit in 2023 is to low as @openshwprojects said but I don't know. I'll dig into that a bit when time permits but I wouldn't expect anything soon on it, sorry.

theyo-tester, 2023-08-19:
I don't think that consumption_stats is the issue here (even if it is not available in HASS. Here, it seems that the device doesn't publish any consumption stats, that's why HASS says 'not available') . I have analyzed the discovery topics published by the device, and there are three extra topics published, which seems to be like placeholders for further sensors...? I don't know their purpose,
Especially: 
<Device_Base_Topic_Name>_sensor_6
<Device_Base_Topic_Name>_sensor_7
<Device_Base_Topic_Name>_sensor_8
They are present at least for my devices: Aubess Wifi Mini Switch 16A with power metering (Chipset BK7231N).

These topics messages have no state_topic info, but they doesn't configure any specific sensor anyway.

I deleted these messages from the discovery topics, and re-introduced the devices in Home-Assistant, by removing and re-adding the mqtt component (caution! when deleting mqtt from HASS, all entities are gone, see below an easier option*).
=> The error did not pop up anymore.  
By now I will avoid to hit the "Start Home Assistant Discovery" again :)

*After deleting the topics, you can also just try to restart Home Assistant or the integration, to see if the error still appears, but I strongly believe, your logger will be silent now :)

davidmonro, 2023-09-04:
This is definitely still an issue.

stefan064, 2024-02-18:
Some of this with missing sensors may be fixed in #1092 , being energy counters that rely on the NTP driver.
Consumption stats may work if represented as multiple entities to break up the data. But I don't use the feature, it seems to overlap with HA's own recorder and statistics. Perhaps consumption_stats is just better off excluded from HA discovery unless stats are enabled.