openshwprojects/OpenBK7231T_App #874: Constant Restart/Reboot and Possible Overheat for 20A SmartPlug BK7231N/CB2S with BL0937

viny182, 2023-07-11:
Hello,

I've recently bought a 20A SmartPlug (link below on *Firmware* section) that uses the BL0937 for power measurement and I suspect that something on the web interface from OpenBeken is causing the CB2S module to overheat which makes the device restart.

I've bought 4 of them, and before flasing OpenBenken they were working fine. After I flashed it (I was dumb and flashed all of them trough tuya-cloudcutter, so I do not have the original backup firmware anymore), they start to restart after some time.

First I noticed that there were huge drops to the voltage being reported to HomeAssistant trough MQTT, like Voltage drops to 60V or even 110V, on a 127V phase.

By looking at the logs I noticed that the drops were happening on the lines for the SSDP module, so I disable it. I notice a slightly improvement on the stability for the restart/drops, but they are still occurring.

Then, I stopped the ping watchdog, to spare some load on the module, to see if any benefits, and the module was stable for a while with no restarts, but still with Power drops. After I couple of time troubleshooting, I notice the drops were happening when I access the Web Interface.

When there is a heavy loads on the device (e;g, > 1400w), and when I access the web interface, the module restarts completely.

When there is no heavy load (<800W, even with 0 load), I see drops but the device does not restart, with SSDP and PingWatchDog Stopped.

I do have other device models using exactly the same chips (BK7231N/BL0937), but they are 10A using lower loads, but even with 800W load they do not send drops on the measurements for HA.

Also, with a multimeter on the devices output, I noticed that the voltage does not drops. Only the measurements are affected when sent to HA trough MQTT, but sometimes as the devices still reboots, it cuts the power to the output, so any appliance connected to it would power off, turning the module unusable. 

By opening the device, I noticed that the power LED is connected in a weird way to the relay, it's not controllable trough BK7231N, it is hardwired to the relay command pin.

I was planning to submit a complete device teardown/review on the forums, but I prefer to solve this issue before.

**Firmware:**
- Version 1.17.174
- Device: **"AVATTO Tuya 16/20A Brazil Standard WiFi Smart Plug Smart Life APP"** ([link](https://www.aliexpress.com/item/1005004988305477.html?spm=a2g0o.order_list.order_list_main.35.21efcaa4Erjq2c&gatewayAdapt=glo2bra))
- Chip/model: BK7231N
- Device config:
Pins: 
6: BL0937CF1
7: BL0937CF
8: Btn
10: WifiLED_n
24: BL0937SEL
26: Rel (with a Loop LED attached, first time I saw it)

Drivers:
SSDP, NTP and BL0937 on startup (SSDP stopped after troubleshooting)

**To Reproduce**
1 - Setup a Power(W)/Voltage(V) monitor tool like HomeAssistant with MQTT. The drops are also visible on the web interface, but it's better to monitor with an external tool as the test requires to open/close it multiple times...
2 - Access WebInterface and click on different links to generate load to observe if there is any dropes on Voltage or Watts
3- Repeat tests with different loads (0W, 800W, 1000W, 1500W)

**Screenshots**
If applicable, add screenshots to help explain your problem.

With Loads, device restarted on second time
[image]

Without Loads, drops are less impacting, but still wrong. I do not have such drops on other different devices...
[image]

**Other relevant info**
I've tried to capture the logs before the device restarts, but to my knowledge there is nothing being logged related to the issue... Apparently it restarts after it can send any useful information to the log screen... Also, with the "debug/all" option activated, and having all the flags checked, there is a lot of output, difficulting the analysis. Is there any tips for what to select to filter considering this scenario?

[image]

viny182, 2023-07-11:
hi @openshwprojects, I've used on one of them, but I did not see any differences to the other 3 devices...

openshwprojects, 2023-07-11:
I know this is a long shot, but powering your device from a reliable 5V power supply (through the 3.3V LDO on the board) could tell us whether the issue is OBK fault or a power supply fault in your particular plug.

Of course, this has to be one in a specific way, otherwise you'll blow up the plug... they are not isolated from mains.

I can see two potential causes of this issue.
1. BL0937 driver relies on an interrupt and maybe it fires too slowly, but... but that would also affect your other plugs you mentioned BK7231N/BL0937
2. your plug has low quality power supply and it falters during wifi transmit/etc....

I also noticed:
_"I've recently bought a 20A SmartPlug (link below on Firmware section) that uses the BL0937 for power measurement and I suspect that something on the web interface from OpenBeken is causing the CB2S module to overheat which makes the device restart."_
I had exactly the same issue in the past and it was caused by not using PowerSave 1 on device which was taking too much current and the tiny Tuya power supply failed and I had to repair it, see topic (use google translate):
https://www.elektroda.pl/rtvforum/topic3898805.html
I was not aware about it back then, but some Tuya devices have so low quality power supplies that PowerSave 1 is mandatory.

viny182, 2023-07-11:
The option 2 totally makes sense...

But then are we supposing that the original firmware uses an option like "PowerSave 1" by default? If so, should we evaluate if OpenBeken should have enabled by default as well, or at least having this option on a place with more visibility to ask user to test before degrading capacitors?

One thing I forgot to mention, I noticed that the LED attached to the command Relay Pin, which is controlled by CB2S PIN 24 does flashes/blink randomly... perhaps there is no sufficient power being provided to the module and I should monitor this pin with a multimeter?

Also, here is a picture of the device... I accidentally damaged the relay and one fuse for this one, but the issue happens for the other 3 which I never opened... And still this one that was damaged is still working like the others hahaha

[image]

viny182, 2023-07-11:
first of all, I don't even know you but I already have a huge respect for you and a deep admiration on all your work.

Asking opinion of a user reporting you a issue, shows character and good personality. Thanks for all you've done for this community!!!!

To decide if we should push "PowerSave 1" in a next commit, I see that none of the users reported issues in the link post you sent, but doing a small risk analysis I would come with this check list:

- "PowerSave 1" can **permanent** harm any devices if set?
- In case of any issues, will be it hard for the user to identify the "PowerSave 1" is the culprit?
- The effort to show its status on the Home WebPage with a disclaimer to the users would be great/huge?

If the answer is "yes" or "maybe" for any of the questions above, then I would **NOT** push it as default in a next commit. But if it's no, then I believe we won't have any major issues with it.

----
The LED seems to be driven with ressitor attached to the command relay... like a pull up resistor maybe? I have not measured it... but I think it will be the next step... In other words, it seems that both Relay Command Pin and the LED are connected in parallel to 24PIN from the CB2S module... (but I still need to better analyze the tracks on the board to see if this is really it...)

----
As for the fix you suggested... I do have a similar capacitor (470uf 16V ) on the board which seems to be fine by visual inspection... I Still need to measure if it is directly connected to any power supply... I do however have also other 2 capacitors of 4.7uf / 400V, and a third one that I will have to take off from the board to read the values... 

To summarize the fix: Do you think that replacing the first 3 capacitors and enable "PowerSave 1" would be something good to try as a good/long shot?

openshwprojects, 2023-07-11:
Thank you for kind words. 

Well, we could only enable PowerSave in WiFi client mode, not in AP mode, but that still wouldn't change fact that beginners will don't know what's happening if PowerSave is breaking their devices as soon as they configure their SSID.

Maybe I could create some kind of popup dialog or a simple main panel display in yellow font saying "PowerSave not enabled - consider enabling!". I will look into that soon.

Regarding capacitors - well, I am afraid that in the case of switch mode power supplies, you have to get a low ESR capacitor rated for switching usage. You can't just get any capacitor out of electronic scrap boards, it has to be low ESR cap. Otherwise you will have to replace it again after a week or after a month. Trust me, I tried.

The cap that breaks is a low voltage one, 16V/470uF or something. No need to replace 400V ones.

Of course, we don't know yet if it's really a capacitor fault. Maybe it's not. 

I will just say that if the LED on GPIO blinks while GPIO is set to constant high (or low) in OBK, then it's bad.

Does that LED on GPIO also blinks if you set GPIO role to "AlwaysHigh" in OBK (or AlwaysLow, choose a proper one)

deltamelter, 2023-07-13:
Oh, I think I am facing the same issue with some new CB2S based plugs.  The biggest problem is not the reboot but the fact that after reboot, the device toggles off instead of keeping the power state. 
Is there any danger enabling powersave 1? I'm willing to try if it will let my device stay on.
Had problems with these connecting to my AP when trying to initially set wifi.

There only appears to be one LED and this gets incorrectly  assigned as wifiLED_n pin, so I have have it changed to an ordinary LED and changed the channel to 1, (same as the button and relay), from 0.   Can this impact in any way?

viny182, 2023-07-13:
@deltamelter in case you don't mind that the whatever device you have connect will turn off/on when the smart plug CB2S restarts, then you can set the relay to be always closed/on in the "Configure Startup" screen, by setting your channel to "1".

But be aware that depending on the load you have, your CB2S module and the device connected to it will restart a lot!

In my case, I have a dishwasher connected to one of the modules, so if I let this happen, probably my dishwasher will fry... So for now, no measurements from the dishwasher, until we figure this out...

[image]

[image]

deltamelter, 2023-07-13:
@viny182  I had this test plug on a chest freezer, I prefer not to have that poweroff randomly with CB2S reboot.  2 of the devices were newer tuya firmware (1.1.12) and the will reboot on their own with no load.
The "older" plug is identical inside with regard to hardware (1.1.8) but seemed to have no problem with the load from the freezer, but when a clothes dryer was turned on, off the same plug, it started rebooting, after an hour at first and later more frequently
Total load was around 1.5KW on the dryer

@DeDaMrAzR  I wondered what this means, mine say:
 Reboot reason: 0 - Pwr

DeDaMrAzR, 2023-07-13:
@deltamelter that would be something related to the power supply, basically something like a power cut.

deltamelter, 2023-07-13:
I didn't calibrate these yet and the voltage very erratic, I will set that first. But I had not interruptions to power...
One of these devices
[image]

On a tasmota powermonitor plug that was calibrated:
[image]

viny182, 2023-07-13:
@DeDaMrAzR, I've got the same reboot status as @deltamelter , 0 - Power.
[image]

-------

Well, I've just got the low ESR 16v/470uf cap that @openshwprojects suggested to replace, and already done the replacement on one of the modules.

As a pre-test, I connected a 2000W hair dryer to it and let it work for 5 min.... The voltage is not very stable (like other devices), and when I've open the web interface I see it dropping to 42v... The LED attached to the Relay on P26 is still blinking/flashing randomly also...

The CB2S device was not hot at all, so I think I'll try to reconnect it to my dishwasher and see if it will turn off during a full cycle .... 

[image]

DeDaMrAzR, 2023-07-13:
@deltamelter I am afraid that what @openshwprojects said it may be the power supply inside of the device it self, not yet sure but  that reboot reason info is somewhat telling. You should get "Reboot reason: 3 - Unk" if it was a FW that is causing those problems. 

Again, not 100% sure, just guessing as I haven't had that issue so far and I've been testing similar devices for 60+ days albeit not with that power supply configuration.

DeDaMrAzR, 2023-07-13:
Also what is the quality of your WiFi signal (RSSI value)? That may impact the consumption as well?

I tried to figure out the topology and what that IC is form the pictures @viny182 provided but so far no luck.

viny182, 2023-07-13:
@DeDaMrAzR , what IC do you need to check? As I have open another module, I see that are some IC than can be easier identified.... Can you screenshot and show me the IC you need?

Here is the Wifi signal value... Should I test it with lower levels as well?

[image]

DeDaMrAzR, 2023-07-13:
@viny182 if you have a disassembled device can you confirm that the topology looks similar to this datasheet? -https://eu.mouser.com/datasheet/2/277/MP150-3106221.pdf

viny182, 2023-07-13:
if you refer to the IC on the pic below, then it is a A1117B 3.3V https://www.alldatasheet.com/view.jsp?Searchword=A1117B&sField=3

(The other bigger one is BL0937)

[image]

viny182, 2023-07-13:
Sorry @DeDaMrAzR , I found the IADGN.... here it is:

[image]

deltamelter, 2023-07-13:
[image]
[image]

DeDaMrAzR, 2023-07-13:
> Sorry @DeDaMrAzR , I found the IADGN.... here it is:
> 
> [image]

Can you check the schematic in the datasheet against your board, does it look similar? I am aiming at it's 200mA nominal current output to blame for your problems (not guaranteed of course - just guessing) 

@deltamelter 

No enthusiastic about getting this datasheet - https://www.sekorm.com/doc/2190585.html

deltamelter, 2023-07-13:
so powersave 1 is it a good idea here?  make any difference?

DeDaMrAzR, 2023-07-13:
> so powersave 1 is it a good idea here? make any difference?

Give it a try but try testing it first, meaning don't use it on something critical that fails with consequences (freezer comes to mind :) )

deltamelter, 2023-07-13:
> Give it a try but try testing it first
currently trying startup value -1 and hadn't noticed it restarted several times until after I plugged in a bulb for load and checked   mqtt...
[image]

I'll try powersave 1 next.  but is this uncalibrated state going to be causing any problems?

deltamelter, 2023-07-13:
I thought online time was resetting at reboot before tho
```
Build on Jul 11 2023 10:05:11 version 1.17.180
Online for 2 hours, 19 minutes and 26 seconds
```

viny182, 2023-07-13:
@deltamelter uncalibrated state woudn't cause any issues... 

but startup -1 means that if device restarts, it will reboot remembering the last relay state... so If you do not have anything connected, your only way to check if it was restarted is looking for the uptime status on the bottom of the web interface.... and if you have set -1, and have a device connected, it will always power on the device if it restarts.... 

Perhaps in a test you should consider setting startup = 0 to see if the device you are connecting gets powered off...

viny182, 2023-07-13:
> > Sorry @DeDaMrAzR , I found the IADGN.... here it is:
> > [image]
> 
> Can you check the schematic in the datasheet against your board, does it look similar? I am aiming at it's 200mA nominal current output to blame for your problems (not guaranteed of course - just guessing)
> 
> @deltamelter
> 
> Not enthusiastic about getting this datasheet - https://www.sekorm.com/doc/2190585.html

Seems very similar to the schematics, but I'm not an expert... is there any specific pin that I should make sure is equal?

deltamelter, 2023-07-14:
-1 works great to keep the plug toggled on, except it does flash off for less than a second.  fine for "analog" stuff where blips off will not be a problem (probably)
PowerSave 1, mixed results so far goes slightly longer between reboots, one device that was rebooting after 10-15mins without load, went over an hour before reboot.  Another device (on the freezer) has not restarted in over 9 hours so far since powersave... 

I had a similar issue with an older tasmota device that started failing to reach the AP after years with no problems.  The only change was a new router/AP. To "fix" that (probably temporarily) I used the minimal tasmota without the bells and whistles and power requirement and it was enough.
This got me to thinking about whether obk could be "minimalised" and I started with turning off NTP service, because I'm not sure why it needs to be running constantly.   Maybe I'm wrong, but I'm not trying to get the plug device to keep its own schedule, just needs to spit out mqtt, does mqtt need accurate times? Does anything else if I am going to use HA to do the "smart" stuff?

Anyway, on the worst rebooting device, last night I ran stopDriver NTP on the console command line and that device has not rebooted since (>9hours so far), there is not load on the device, but that didn't stop it rebooting before....   Will put some load on this and test a bit longer.

What else can be "minimalised"/"dumbed-down" on the device to reduce the power needed?

deltamelter, 2023-07-14:
[image]
over11hrs and rebooted...
see the top for the difference after stopping NTPtho

deltamelter, 2023-07-14:
I only stopped NTP on commandline, not in the startup command so after the reboot, it's has rebooted again twice...

viny182, 2023-07-14:
Hi, I confirm that stopping the NTP driver gives it more stability... Just like stopping the SSDP driver like I mentioned on the first post... I think we are on the right track... Now we should investigate what else can be disabled...

DeDaMrAzR, 2023-07-14:
To my knowledge (not that I have too much of it) NTP is used for consumption_stats that are problematic in their own way as the implementation of it is incompatible with HA due to the state limit, @openshwprojects is aware of that, and I just did some quick test to check the current consumption and can confirm that enabling that driver (NTP) caused a spike of almost double the nominal current (from 90 to 170+mA). 

So I guess disabling NTP and enabling Powersave can be a possible solution?

viny182, 2023-07-14:
@DeDaMrAzR can you try to disable EnergyStats to see if there is any relevant consumption drop? (SetupEnergyStats 0 60 60).

I've already disable it on here, and guess what: More stability....

We can live without it as we can use Riemann Calc helper from HomeAssistant to generate the daily/total historical statistics...

DeDaMrAzR, 2023-07-14:
@viny182 I am having trouble testing with my socket (AC danger) but will test soon and report back. It takes some time to set that up so please be patient - but by the first test there is a difference, both of you are on the right track and power supply is 100% to blame @openshwprojects is correct in that aspect. 

I'll try testing and recording on my test boards as that may be easier but the results may be interesting as well. 

Thank you both for you effort to make this project better!

DeDaMrAzR, 2023-07-14:
This is that short test only on a test device (not a plug with power meter)!

[image]
[SPDXXXX_data.csv](https://github.com/openshwprojects/OpenBK7231T_App/files/12052221/SPDXXXX_data.csv)

openshwprojects, 2023-07-15:
It looks like a really good piece of research was done here. Good job! I am surprised by that NTP power requirement peak, but our NTP relies on Beken libraries so that requires more investigation.

In a meantime, could anyone with problematic plug try disconnecting them from mains and **powering the from 5V at the 3.3V regulator input** to see if problem persists? Of course, use a good 5V power supply.

viny182, 2023-07-15:
**IMPORTANT UPDATE:**

I run 15+ openbeken devices, and some of them were no updated to latest version, so I decided to updated all of them except one that was not being used (even open another post on how to do it [link](https://github.com/openshwprojects/OpenBK7231T_App/issues/875)), and after I updated them, 3 or 4 devices that were stable started to reboot as well!

So, I decided to downgrade all my devices to the FW version from the one that I have not updated **(1.17.58)**, and the Voltage measurements are way more stable, even for the 20A device that started this issue... So now maybe we have something to investigate between this 1.17.58 version and the newer ones??

Picture below for the measurements, still not calibrated yet but it does not variates like in the newer FW that was installed before... Also, I still need to test it with high loads to see if it restarts...

PS: The downgrade made me lost all configuration for the devices, so I had to setup it from scratch....

[image]

deltamelter, 2023-07-15:
I have 8 plugs that mostly failed to flash wirelessly, probably because the wifi and power was sketchy, I have 1 on 1.7.175, 4 on 1.7.180 and the rest still to be flashed via serial.  serial is not easy with these because the useful pins can only be accessed "blindly" and not sure where I would need to connect stable power.  If to the CB2s 3v3 pin, it would be hard to do consistently.

Can I downgrade via OTA interface? or would I have to try serial flash?

deltamelter, 2023-07-15:
also, what's the best way to backup all of the config to apply on multiple devices?

DeDaMrAzR, 2023-07-15:
@deltamelter as @viny182 mentioned you will lose your config if you downgrade so be aware of that!! Make screenshots, download templates, config files and autoexec before that!! 

Downgrading is not advised due to multitude of reasons and you have to understand that the project is in rapid development and **feedback from both of you is very appreciated.** If you are do downgrade via OTA do it in small increments, version by version (again NOT advised!!)

Trouble is that there is no way to test and follow up on all devices and potential problems with them and especially if the device is not in our hands and we can't recreate the problem on the devices that we own.

@viny182 great news that you found the build that works for you, can you "sacrifice" that device now and play with it by updating versions slowly to see which build will start behaving problematically? (1.17.58 is almost 130 builds "old")

@deltamelter try using the flash tool - https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.1.1

deltamelter, 2023-07-15:
@DeDaMrAzR I'm only looking to try the older fw on one plug to test if it really makes a difference.  Only thinking of downgrade to get there so I can test like for like.  All the other plugs are hardware identical, so I could flash directly.

> @deltamelter try using the flash tool - https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.1.1
I used this tool to serial flash 5 of the plugs already.  Another 1 is tuya-cloudcut original FW, 1 in cloudcut flashed to obk 1.7.175.  4 were patched tuya and had to be serial flashed and they are all on 1.7.180 and least stable

When talking about backing up and copying configs, I would only be looking to apply these to the same FW version, to update all the plugs once I have settled on a "working" setup :smile:

viny182, 2023-07-15:
I've downgraded by OTA... Lost all the configs, but at least ALL my 15 devices booted on AP mode, so I did not need to flash trough serial....

I'll choose 1 device to update gradually to try to identify which is the "faulty" version and let you guys know.... 

But as there is 100+versions to test, it will take some time... I'll made some jumps between the versions to try to narrow the range with issues... But anyway, Not sure if I'll be able to make it this weekend because of some other appointments I have....

Once again, thank you all for the help and support!

deltamelter, 2023-07-15:
> I've downgraded by OTA... Lost all the configs, but at least ALL my 15 devices booted on AP mode, so I did not need to flash trough serial....

OTA from 181 to 58?

viny182, 2023-07-15:
> > I've downgraded by OTA... Lost all the configs, but at least ALL my 15 devices booted on AP mode, so I did not need to flash trough serial....
> 
> OTA from 181 to 58?

Yes!

openshwprojects, 2023-07-15:
I am sorry to hear that you've lost the config. Unfortunatelly, we had to add some more config space for second SSID field, etc, so backwards update from v4 config to v3 is not supported. We do not, however, plan to alter flash config any more so most likely future versions backwards OTA will work correctly.

I never recommend mass updating devices, it is always worth to check the operation on one first.

If you suspect that something changed in the firmware, please check which version is guilty.

At about 100 versions to check, you should use the bisection approach. Start at the middle, let's say , ver 50, then see if it breaks. If it breaks, then check ver 25, if not, check 75, at so on.

Not much was changed for power handling these days, here is a log file for bl_Shared :

https://github.com/openshwprojects/OpenBK7231T_App/commits/main/src/driver/drv_bl_shared.c

On  May 2 the simple configurable floating point rounding was added, but could that be the culprit? 

Of course, the issue could be in the another file...

deltamelter, 2023-07-15:
This device was flashed to 1.7.180 but not yet configured, OTA downgraded to 1.7.58
set wifi, started BL0937 manually and added mqtt broker and set startup to -1.  Also set voltage
lasted longed than the other device, but still rebooting
[image]

deltamelter, 2023-07-15:
This device is running 1.7.175, PowerSave 1, mqtt and startup -1
Has been going >24hrs now without reboot.   This one has been under (light) load and a few periods of heavier load.
This has identical looking hardware but was older date and older tuya firmware, was flashed with tuya-cloudcutter, that's the only (obvious) difference.
[image]

deltamelter, 2023-07-16:
The (freezer) plug that I set PowerSave 1, turned off NTP and consumption stats, running 1.7.175 has not rebooted in coming up for 48!! hours now... I have rolled 2 other plugs back to 1.7.175 to similar effect so far...
[image]

[image]

1.7.58, is better than 1.7.180, but still restarting for me....

deltamelter, 2023-07-17:
Is there a way to flush the config and storage and just OTA flash new firmware?
2 plugs that were either serial flashed straight to 1.7.175, or to 1.7.180, then OTA to1.7.175 are still going almost 3rd complete day without rebooting 

> Build on Jul 8 2023 09:02:55 version 1.17.175
> Online for 2 days, 18 hours, 6 minutes and 4 seconds

Other 2 plugs that were configured with energy stats, NTP and mqtt where later downgraded from 1.7.180 to1.7.175 (one via 1.7.58) and stopped NTP, PowerSave 1, disabled energystats and retained config between versions, they keep on rebooting like crazy.  Not as much as with 1.7.180,and going longer sometimes in between reboots, but still rebooting, sometimes with just a couple of mins in between.

viny182, 2023-07-17:
@deltamelter I have not tested this, but take a look in here: https://github.com/openshwprojects/OpenBK7231T_App/issues/500

TL;DR: There is a "ClearConfig" command.

viny182, 2023-07-17:
there is also a clearAll command:

<html><body>

clearAll |   | Clears config and all remaining features, like runtime scripts, events, etc | File: cmnds/cmd_main.cFunction: CMD_ClearAll
-- | -- | -- | --
clearAllHandlers |   | This clears all added event handlers | File: cmnds/cmd_eventHandlers.cFunction: CMD_ClearAllHandlers
clearClockEvents |   | Removes all set clock events | File: driver/drv_ntp_events.cFunction: CMD_NTP_ClearEvents
clearConfig |   | Clears all config, including WiFi data

</body>
</html>

deltamelter, 2023-07-17:
Don't know what else is different between these 2 devices...

[image]|[image]

deltamelter, 2023-07-17:
Nothing got "Fried" 😀 
needs a reboot?  it will do that itself any second...

```Info:CFG:CFG_SetDefaultLEDCorrectionTable: setting defaults
Info:CMD:Fried 0 handlers
Info:CMD:Fried 0 rep. events
Info:CMD:CMD_ClearAll: all clear
Info:CMD:[WebApp Cmd 'clearALL' Result] OK
Info:MAIN:Time 124, idle 147995/s, free 61608, MQTT 1(1), bWifi 1, secondsWithNoPing 53, socks 3/38 POWERSAVE
Info:MAIN:Time 125, idle 75232/s, free 73304, MQTT 1(1), bWifi 1, secondsWithNoPing 54, socks 2/38 POWERSAVE
Info:MAIN:Time 126, idle 72306/s, free 73304, MQTT 1(1), bWifi 1, secondsWithNoPing 55, socks 2/38 POWERSAVE
Info:MAIN:Time 127, idle 76897/s, free 73304, MQTT 1(1), bWifi 1, secondsWithNoPing 56, socks 2/38 POWERSAVE
Info:MAIN:Time 128, idle 76785/s, free 73304, MQTT 1(1), bWifi 1, secondsWithNoPing 57, socks 2/38 POWERSAVE
Info:MAIN:Time 129, idle 74210/s, free 73304, MQTT 1(1), bWifi 1, secondsWithNoPing 58, socks 2/38 POWERSAVE
Info:MAIN:Time 130, idle 82263/s, free 49912, MQTT 1(1), bWifi 1, secondsWithNoPing 59, socks 4/38 POWERSAVE```

deltamelter, 2023-07-18:
`clearAll` seemed to help, but the 2 devices that restart continuously still do so, they seemed to go for 24 hours without rebooting, but now back to their instable ways...  
Any chance of a debug utility to see what is running, taking up processing resources and therefore power?
like an `obkTop` command :)

openshwprojects, 2023-07-20:
I am not sure currently if it's possible to add such feature.

How's the devices stability after 2 days?

@deltamelter can you tell me whether those devices restart when run with clear config (no drivers, no power metering, etc)?

deltamelter, 2023-07-20:
> @deltamelter can you tell me whether those devices restart when run with clear config (no drivers, no power metering, etc)?

After ClearAll I started power monitoring and powersave 1.  These 2 devices didn't improve.  One of those is currently running libretiny-esphome to see how it performs, the other I unplugged because the constant click-click was starting to get annoying :grinning: 

One of the "older" devices has been running happliy on 1.7.175 with BL0937 loaded and mqtt for nearly 6 days
`Build on Jul 8 2023 09:02:55 version 1.17.175
Online for 5 days, 22 hours, 17 minutes and 30 seconds` 

The other one on 1.7.175 (down from unconfigured 1.7.180) is currently unplugged, but had not rebooted on its own.

I will plug in one of the less stable devices and clearAll again and leave it with no drivers, should I at least configure the wifi?

viny182, 2026-05-02:
Hey,

Sorry for bumping this old thread... But I got a clue about this issue now...

Back at the time I have reported, I had 2 routers with the same SSID... (2 routers with same network name).

I have turned off one router some months ago and did not turned it on ever since... Today I started to play with one devices that was constantly rebooting and to my surprise it seems to be stable now...

Maybe having two networks with same name was the issue? If anybody of you guys have a network setup like me, that could be a valid test...