openshwprojects/OpenBK7231T_App #875: [QUESTION] How to update and manage multiple devices OTA?

viny182, 2023-07-14:
Hello all,

I currently have 15+ devices running OBK, and I was wondering if there is an easy way to update them all at once, instead of going to each device and updating them manually...

I see we have available the command "ota_http", which requires a local http server.

Is there a way to send this command trough MQTT to all devices at once to make them update at bulk? Of course I would do it only after testing with at least one device....

viny182, 2023-07-14:
I think I figured this out... 

If I send the command to the Group Topic, all devices will reply.. here is how I've done with home assistant Mosquitto integration and local http server:

[image]

[image]

closed 2023-07-14 by viny182

ataa, 2023-07-14:
# pyOpenBeken
An easier way to manage your [OpenBeken](https://github.com/openshwprojects/OpenBK7231T_App) devices.

The purpose of this project is to ease the control of multiple devices avoiding all manual tasks.

### Features
* Send commands
* Setup mqtt & HA autodiscovery
* Read file system
* Push OTAs and restore FS

### Planned features
* Scan for devices
* Set pins (Roles, Channels)
* Control device
* Set other options

https://github.com/gargomoma/pyOpenBeken

openshwprojects, 2023-07-14:
Please remember to separate T and N!