openshwprojects/OpenBK7231T_App #876: Unreachable address after inernet restart

Jonsonmon, 2023-07-15:
After internet restart, the device is connecting to the wifi (I Can see it on my router connections, its there!)
But cannot ping to the device IP (Unreachable IP) . only physical touch is working.
Disconnect the device from the router for few seconds is fixing the problem.
Cannot reach the device then I Cant access to logs....

**Firmware:**
- Version [e.g. 1.14.129] 1.17.181
- Device?  AVATTO Wall Smart Switch 1 GANG
- Chip/model: BK7231T? BK7231N
- Device config?

openshwprojects, 2023-07-15:
Can you specify whether device is T or N? You can find it out on the main WWW panel of obk.

Jonsonmon, 2023-07-16:
> Can you specify whether device is T or N? You can find it out on the main WWW panel of obk.

I'm using BK7231N .
I'm using DHCP Reservation (Static IP) in my router...

I also see this:
boot incompletes 1 (might change to 0 if you wait to 30 sec)!
This may be linked to my problem?
tried to chane to 30, but still same message.