openshwprojects/OpenBK7231T_App #877: Incorrect mask when setting fixed ip

bartsnijder, 2023-07-15:
FW version:1.17.182

When setting a fixed ip, I set the mask to be 255.255.255.0, but when applying it is set to 255.255.0.0. Only when setting the mask to 255.255.255.0 for the second time, it is properly set.

openshwprojects, 2023-07-15:
Hello, do you reboot device after saving IP configuration?

bartsnijder, 2023-07-16:
The change from 255.255.255.0 to 255.255.0.0 already happens in the web interface after I press "submit". The change becomes only affective when reboting obviously (but mask 255.255.0.0 is then indeed set)

DeDaMrAzR, 2023-07-16:
@bartsnijder can you try doing that setup in incognito mode or another browser maybe? 
I tried on multiple devices (both N and T with various FW builds) but couldn't replicate the error so I am suspecting something about your browser cache may be the problem.

openshwprojects, 2023-07-16:
I tested on BK7231N and on Simulator and was no able to reproduce.