openshwprojects/OpenBK7231T_App #878: Configure module with fixed IP

bartsnijder, 2023-07-15:
FW version:1.17.182
After setting a fixed IP addess, opening the "configure module" section page stalls and the device disconnects from wifi. Rebooting /repower gets recovers the device.
When setting IP back to 0.0.0.0 (dynamic), "configure module" works again

openshwprojects, 2023-07-15:
Hello, as far as I remember, you need to reboot so the static IP change can take place. Still, I will test it myself tomorrow on my board and we will see if it works.
Btw , are you using T on N module?

Jonsonmon, 2023-07-15:
> 

I'm using BK7231N .
I'm using IP Reservation (Static IP) in my router...

Jonsonmon, 2023-07-15:
I See this:
`boot incompletes 1 (might change to 0 if you wait to 30 sec)!`
This may be linked to my problem?

bartsnijder, 2023-07-16:
> Hello, as far as I remember, you need to reboot so the static IP change can take place. Still, I will test it myself tomorrow on my board and we will see if it works. Btw , are you using T on N module?

Forgot to mention that I do reboot in between. So With the web interface set to the new fixed IP, the issue occurs. The units where it currently happens is: BK7231N

bugalaws, 2023-07-16:
I got the same problem too, using in bk7231n too