openshwprojects/OpenBK7231T_App #882: How can I automatically publish cmnd/OBK_DEVICE/POWER via MQTT?

CLAM01, 2023-07-24:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version: 1.17.199
- Device: LSC Mood Light
- Chip/model: BK7231N

**Question**
I have successfully powered my LSC Mood Light with OpenBK via the flasher.

After I found out that only the information of the single states like Color, Dimmer etc. are published via MQTT, I try that the cmnd states are published automatically.

I use IOBroker and it is tedious to always create all data points manually, especially when you are in tests.

Is there a way to publish the cmnd states automatically?
Tasmota is very user friendly at this point.

openshwprojects, 2023-07-24:
Hello, if you are using ioBroker, have you checked our flag to enable Tasmota TELE publishes? Have you checked flag to broadcast self state automatically? We have a YT tutorial for that:
https://www.youtube.com/watch?v=x4p3JHXbK1E&ab_channel=Elektrodacom

CLAM01, 2023-07-24:
Hi, actually it is. But I may have done something wrong.
I'll have a look at it.

CLAM01, 2023-07-24:
OK, I have now used another MGTT server in IOBroker. So the Sonoff adapter. There I can automatically create the states for stat, TELE etc.. Everything comes in. Top and many thanks to you.

CLAM01, 2023-07-25:
I miss the IP Adress off device on MQTT with flag for Tasmota stat.

openshwprojects, 2023-07-26:
Hello, which STATUS command are you referring to and how the JSON should look like?

CLAM01, 2023-07-26:
Ich verstehe die Frage nicht ganz.

Aber als bei einem openBeken bekomme ich dieses JSON per MQTT:

Da hätte ich neben der SSID des WLAN auch meine IP erwartet.

`Client [MoodLight_Kueche_Kuehlschrank] Resend message topic: tele/MoodLight_Kueche_Kuehlschrank/STATE, payload: {"Time":"1970-01-01T00:00:00","Uptime":"1T09:44:28","UptimeSec":121468,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":23,"Dimmer":29,"Fade":"OFF","Speed":1,"LedTable":"ON","Color":"0,0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0],"CT":500,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SIMMER02","BSSId":"xx.xx.xx.xx.xx.xx","Channel":11,"Mode":"11n","RSSI":102,"Signal":-49,"LinkCount":21,"Downtime":"0T06:13:34"}}`

Bei Tasmota bekomme ich aber auch noch so ein JSON:

`[DVES_C15940] Received: tasmota/discovery/A848FAC15940/config = {"ip":"192.168.2.240","dn":"Tasmota","fn":["Tasmota","Tasmota2","Tasmota3",null,null,null,null,null],"hn":"tasmota-C15940-6464","mac":"A848FAC15940","md":"Generic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.2.0","t":"tasmota_C15940","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1}`

Ich hoffe das hilft?

openshwprojects, 2023-07-28:
Under which topic is the followinng JSON published?

CLAM01, 2023-07-28:
```
14:10:13.075 MQT: tele/tasmota_C15940/INFO1 = {"Info1":{"Module":"Generic","Version":"12.2.0(sensors)","FallbackTopic":"cmnd/DVES_C15940_fb/","GroupTopic":"cmnd/tasmotas/"}}
**14:10:13.083 MQT: tele/tasmota_C15940/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-C15940-6464","IPAddress":"xxx.xxx.xxx.xxx"}}**
14:10:13.095 MQT: tele/tasmota_C15940/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":165}}
```

This is from my Tasmota Device Log..

And its store in INFO on my IOBroker

[image]