openshwprojects/OpenBK7231T_App #883: Value "true" for cmnd POWER is not recognized

CLAM01, 2023-07-24:
**Describe the bug**
I tried using the cmnd to turn my outlet off and back on with the command "http://<ipAddress>/cm?cmnd=POWER%20<true/false>".

With the value "false" the socket is also switched off.
If I send the command "true", nothing happens.

I integrate the socket into IOBroker and there the bool values are always written with "true" and "false" into the IOBroker atenpoints.

Is this a bug or rather a feature request?

**Firmware:**
Version: 1.17.199
Device1: LSC SMART Plug
Chip/model: BK7231T

Device2: LSC Mood Light
Chip/model: BK7231N

**To Reproduce**
Steps to reproduce the behavior:
1. Switch POWER on
2. Send command "http://<ipAddress>/cm?cmnd=POWER%20false"
3. See the Light oder Plus is off
4. Send command "http://<ipAddress>/cm?cmnd=POWER%20true"
5. See, nothing powered on

openshwprojects, 2023-07-24:
Hello, where is true used? As far as I know, only 1, 0, TOGGLE, ON and OFF are used in Tasmota. 

I can add "true" for you, but I need to know if it's really used anywhere.

CLAM01, 2023-07-24:
No, especially for me it does not have to go in.
Only if it would be generally desired from the community.

openshwprojects, 2023-07-25:
My question is, is "true" used in any system with Tasmota? Ddoes Tasmota support "true"?