openshwprojects/OpenBK7231T_App #884: Switching 2nd Channel on an LED channel via double press does not work with activated flag 18

CLAM01, 2023-07-24:
**Describe the bug**
When the flag 18 is activated on my Mood Light, it no longer works that the 2nd channel that is set under the configuration is switched on when double-clicking.

Here the channel flashes briefly and goes directly off again.

If I deactivate the flag 18 again, the 2nd channel also works via a double click of the power button.

**Firmware:**
Version: 1.17.199
Device1: LSC Mood Light
Chip/model: BK7231N

openshwprojects, 2023-07-24:
Hello, can you provide full configuration of your device, so I can try?

As far as I remember, the button with two channels is for relays, it is not implemented for LEDs. for Leds we have "BtnSmartLED".

openshwprojects, 2023-07-24:
Your device is a LED. Second channel is for relays. How would you like the second channel to work for LED?

Second channel is only used when you have a two relays device and want to connect them with single button.

CLAM01, 2023-07-24:
OK.
But as long as the flag is not set, double-clicking the second channel works fine.
Therefore I thought that it should be so.
It would be great to be able to use this even with the flag set.

Without the flag "soft transition" it works fine.

openshwprojects, 2023-07-25:
Double clicking does not work correctly, because it sets raw value of channel, not the value in the LED driver. Smooth lighting breaks it because it applies channel value every 50 ms or so to create smooth effect.

How would you like secondary channel to work for LEDs? I can implement it for you, but I need to know what is the expected behaviour.