openshwprojects/OpenBK7231T_App #890: RGBW support for DDP

dg98, 2023-08-02:
I use WLED's DDP integration to control some RGBCW lights running OpenBeken. Most of the problems I had with DDP support were fixed in #690 but no matter what I send in WLED, the white channels remain at 0. 
In [this](https://github.com/openshwprojects/OpenBK7231T_App/issues/690#issuecomment-1445030996) comment you said byte 2 was always zero using xLights, but WLED should set byte 2 to 0x1A as can be seen [here](https://github.com/Aircoookie/WLED/blob/3270605b4f401acc16d42262c81ecffe89647df3/wled00/udp.cpp#L727C59-L727C59) and [here](https://github.com/Aircoookie/WLED/blob/3270605b4f401acc16d42262c81ecffe89647df3/wled00/src/dependencies/e131/ESPAsyncE131.h#L56), which sets the data type to 011 (RGBW) as listed in the screenshot from the DDP spec you added in your comment.

openshwprojects, 2023-08-02:
Indeed, currently DDP parses only RGB:
[image]
So basically we need to add IF here to check for 0x1A and then treat as RGBW? What is the format of RGBW, then? RGBCW or RGBW only?

dg98, 2023-08-02:
> So basically we need to add IF here to check for 0x1A and then treat as RGBW? 
I think so, although it was [changed](https://github.com/Aircoookie/WLED/blob/7d84de6690016cb33055a89d1017c54209b40ef7/wled00/src/dependencies/e131/ESPAsyncE131.h#L57) to 0x1B in 

> What is the format of RGBW, then? RGBCW or RGBW only?
If I understand correctly, it's RGBW only but I'm trying to find out more. Maybe [this](https://github.com/Aircoookie/WLED/issues/3207#issue-1716407895) helps.

openshwprojects, 2023-08-05:
RGBW as in RGB + Warm White? Or RGB + Cool White?

I can implement it right away, I just need to know the meaning of data.

dg98, 2023-08-05:
Is it possible to add a debug output to the log that shows the DDP packet contents? Or share the DDP receiver you used in the other issue? Then I could easily test what WLED is sending in the white channel

openshwprojects, 2023-08-06:
Yes, no problem, maybe something like that?
https://github.com/openshwprojects/OpenBK7231T_App/pull/893

can you get that build and check?

BTW: please send just one packet, otherwise many prints per second will kill OBK

dg98, 2023-08-07:
I just tried, this is an example log entry:
``Info:DDP:DDP data: type=27 r=255 g=255 b=255 w=255``
The type was 26 (0x1A) before I updated to the latest WLED version and now it's 1x1B, so the SSS part of the value reflects 8 bits per pixel instead of 4. Don't know if that matters here.

This is with the color set to white and WLED's white channel set to the maximum. There also exists a color temperature slider that doesn't affect the w value.
These are entries with the color set to the warmest and coldest settings:
C: ``Info:DDP:DDP data: type=27 r=135 g=167 b=255 w=255``
W:``Info:DDP:DDP data: type=27 r=255 g=41 b=0 w=255``

openshwprojects, 2023-08-07:
So if type is 0x1B, then we have 4 values? 

Okay, then how should I use that W value? Is it the value for cold white or warm white?

Also... does it mean that RGB should be on together with white? 

I am asking because most of Tuya devices are designed in such a way that power supply cannot handle turning all channels at once. Tuya didn't design products for that. In Tuya firmware, you either have RGB on or CW on.

dg98, 2023-08-07:
> So if type is 0x1B, then we have 4 values?

Yes, if I set WLED to send RGB the value is 11 and with RGBW it's 0x1B.

> Okay, then how should I use that W value? Is it the value for cold white or warm white?

I'm not sure, do you think it's possible to calculate a CCT value from these RGB values?
If not, maybe you could just use the white channel as the brightness value and a trigger to enter temperature mode and use the led_temperature setting that can be set through the UI?

> Also... does it mean that RGB should be on together with white?

I don't think it should be on, it should switch the temperature mode that's entered when the temperature slider in the UI is changed imo.

openshwprojects, 2023-08-07:
So basically something like that?

https://github.com/openshwprojects/OpenBK7231T_App/pull/893/files#diff-5948280d83b255b04dc45f24304a214e632321cc84abe61133f935aeed09b269

Ok now we need to do the calculations part.
You have posted:
```
C: Info:DDP:DDP data: type=27 r=135 g=167 b=255 w=255
W:Info:DDP:DDP data: type=27 r=255 g=41 b=0 w=255
```
Why is W here 255 all the time? What does W represent?

dg98, 2023-08-07:
> So basically something like that?

Looks fine

> Ok now we need to do the calculations part.

What kind of calculations do you have in mind? Something like [this](https://kno.wled.ge/features/cct/)?

> Why is W here 255 all the time? What does W represent?

W represents this "white channel" slider. I guess it should represent the brightness of the currently selected color temperature?

[image]
[image]

tech1977, 2024-12-03:
hi, I'm trying DDP protocol on this bulb:
https://www.elektroda.com/rtvforum/topic3969672.html
I'm very interested in DDP RGBW support, are there any updates?