openshwprojects/OpenBK7231T_App #892: EV charger scenario : max power or time since switched on

analystcmyk, 2023-08-06:
I have the following scenario. 

I have an EV (electric vehicle) charger that I allow for public use:  if somebody asks for it, I will switch it on..
But to keep it reasonable, I would like to both limit the power and the time.
So I would need to be able to measure the power SINCE switch on event. 

**Setting**: 

The charger is outside, and is just a dumb charger (not smart)
I am using a DIN switch/powermeter with openBK that is controlled by MQTT. 

**Behaviour**: 

Then I would like to switch it off when either : 
- the amount of energy used is more than .....  KWh 
- the time using the charger is more than .... minutes.

Looking at https://github.com/openshwprojects/OpenBK7231T_App/issues/676 I could imagine a command like 

`addSingleRunEvent maxRunTimeMinutes 90 maxRunTimePowerUsedKWh 20 ToggleChannel 0;`

which would switch channel 0 off when either 90 minutes have passed or more than  20 KWh would have been consumed. 

Apart from that it would be nice to have the runtime and power since switch on would be available via MQTT.

Any suggestion on how to do that with OpenBK? 

The config has no home assistant or whatever, only MQTT

openshwprojects, 2023-08-06:
Hello, what kind of meter type do you have? TuyaMCU based one or a generic one?

I think the simplest and most user-friendly solution here would be just adding a small driver doing what you described. It could be enabled and configured when needed. I can do that for you.

analystcmyk, 2023-08-06:
Currently I have access to either
EAMPDW-TY63 ( https://www.elektroda.com/rtvforum/topic3959907.html )
or 
Tomzn TOML-VA (https://www.elektroda.com/rtvforum/topic3975610.html )

Both are Tuya, running openBK, so generic Tuya would be preferred I think because most of what you buy online would be similar. 

I would like to limit the needed backend to mqtt broker only.

At the end I think that having a simple way to share your charger fairly (infrastructure, collective) would be preferred above investing in larger (individual, owner only) battery packs.

openshwprojects, 2023-08-07:
It seems like a very interesting and nice idea, it will be my pleasure to help with that.

So, how do you have power KWh stored? It's in a channel? What's the channel type?

Is it:
`setChannelType 5 EnergyTotal_kWh_div100`  ?

analystcmyk, 2023-08-07:
Yes, correct, 
`setChannelType 5 EnergyTotal_kWh_div100`

I would like t to use the TOMZN TOML-VA ( https://www.aliexpress.com/item/1005005219437317.html ) because it has an additional overtemperature protection.

FYI, a lot of low range chargers use this controller : https://www.etek-electric.com/products/electric-vehicle-charging-station/epc-ev-charging-station-controller/ekepc2-c-s.html

openshwprojects, 2023-08-07:
I will prepare a driver for you. It should be small enough to be included in main build. 

Ok, what about the "power loss" situation? Do we support it (it's complicated), or do we assume that power loss just resets charging?

analystcmyk, 2023-08-07:
I think after a power loss you would to have it switched off and keep it switched off.

Power loss could indicate something may have tripped due to overload or any other issue. 

Current limiting on these devices act in a reset fashion (eg overload = switch off, and it switches on again after a timeout, eg 30 secs. ) but I think openBK in "EV charger mode" should stay switched off imho.  

You may consider to make it an additional parameter, but that might conflict with backlog ( power on relay state ) 

BTW if you look for a non smart charger produced in EU (Poland) then have a look at https://doktorvolt.pl/en/
because their design allows for adding the DIN powermeter _inside_ the same charger.

openshwprojects, 2023-08-08:
I have submited a draft of a driver for you, but I will also write a self test for that driver soon:
https://github.com/openshwprojects/OpenBK7231T_App/commit/30aa47f4f74c689bb534ad91cc30cfb5d590fa85

openshwprojects, 2023-08-10:
I am finalizing this driver, can you check if it works correctly? I will also try to improve the UI display.

We could even add a progress bar, but everything takes time and there are hundreds things to do in this project, hmm.

analystcmyk, 2023-08-11:
Sorry to ask but that would mean i would need to compile the binary myself to test? I dont have the toolchain set up currently. Id be happy to flash the adapted firmware and do the tests.

analystcmyk, 2023-08-11:
Ok, Flashed  and looks nice, good work  !

I like the fact that you may use any channel as a limit, eg temperature. 
This makes it very powerful as a limiter for any channel !

I also like that you can disable the timeout.

_Few remarks :_ 

- `[maxAllowedLimitChannelDelta]` `[timeoutOr-1]` should be start/reset when the channel goes from **Off to On** state. ( that is when we explicitly start charging )
Currently it starts counting down after reboot even when it is switched off. 

- When the limit is reached, keep the last values in the display to get an idea how much electricity was used or what triggered the "POWER OFF" :  time limit reached  or delta limit reached for the channel. Only by switching it on again we can reset it. 

- Suggestion : since we do not know what the channel actually measures, we may consider adding a "limitChannelUnit" string var to chSetupLimit () in order to display it in the GUI, eg `limitChannelUnit` =  `Wh` would display the delta as **2522/3500 `Wh`** 

see also https://github.com/openshwprojects/OpenBK7231T_App/commit/30aa47f4f74c689bb534ad91cc30cfb5d590fa85#r123899196

I think a progress bar is out of scope ;)

openshwprojects, 2023-08-11:
I will look into it today or tomorrow. The first remark looks strange. I think it should work if you plug it into: addChangeHandler Channel1 == 1 do_smth_here

The third remark could be done more automatically by sampling channelType that is set for target channel.

Btw, that whole charger idea is very interesting. If you are doing some kind of initiative, feel free to write a few words about that on our Tutorials section (post some photos of your setup if you want, maybe post info about OBK script when it's ready):
https://www.elektroda.com/rtvforum/forum517.html
For a more detailed summary, we can send you a gift: https://www.elektroda.com/rtvforum/topic3950844.html

analystcmyk, 2023-08-12:
Ok tried this :  

- enable or disable the ChargeLimit driver. (suppose I want to charge my own car, then no limit should be applied)
- still not working as expected : POWER OFF should stop the timers/counters and keep the last values  on screen, and only reset the session when toggling.

My first attempt in autoexec.bat : 

`
// Charger mode 
// missing : how to set the GUI name of toggler (channel 29) and label for CH30,CH31 ?
// We use toggle 29 to enable/disable the charger mode
// while enabling, we start a new cycle immediately

setChannelType 29 Toggle
addChangeHandler Channel29 == 1 startchargermode
addChangeHandler Channel29 == 0 stopchargermode

// we keep the maximum time in channel 30 ( $CH30)
setChannelType 30 TextField

// we keep the maximum power in channel 31 ( $CH31)
setChannelType 31 TextField

// (re)start session when toggled
addChangeHandler Channel1 == 1 startsession  

alias startchargermode startDriver ChargingLimit echo "driver ChargingLimit is started" startsession  
alias stopchargermode stopDriver ChargingLimit echo "driver ChargingLimit is stopped" "POWER OFF"
alias startsession  chSetupLimit 1 $CH30 $CH31 "POWER OFF"
`

Ill think about the Tutorial, thank you for asking ;)

openshwprojects, 2023-08-13:
i've changed driver handling so it keeps values on screen, but you may want to alter the way they are displayed. For example, should it show how long is currently charging, or how much time is left till the time limit?

Your toggle script is nice, but I assumed that you will just link it to the toggle of main relay, so the limit is started automatically. Still, it makes sense that you want to have a separate control over the limit...

Maybe it would be easier to use httpButtons driver to add just some buttons like "Start charging limit [bike]", or "Stop charging limit", etc? 

```
startDriver httpButtons
setButtonEnabled 0 1
setButtonLabel 0 "Turn off after 10 seconds"
setButtonCommand 0 "addRepeatingEvent 10 1 Power Off"
//setButtonColor 0 "#FF0000"

setButtonEnabled 1 1
setButtonLabel 1 "Set Red 50% brightness"
setButtonCommand 1 "backlog led_basecolor_rgb FF0000; led_dimmer 50; led_enableAll 1"
//setButtonColor 1 "blue"
```

analystcmyk, 2023-08-13:
Yep definitely better, of course I did not even know about the httpButtons driver ;)  Thank you for the hint. 

And indeed, the separate control is there to use the charger without any limit. 

Will do an OTA and feedback.

closed 2023-09-09 by analystcmyk

jkwim, 2023-12-19:
Very interesting project!

I am doing something in line for my Home EV Charger but not really for public use.

I only do ON/OFF control based on some MQTT messages flowing through my MQTT network from other devices eg. My off grid solar system.

I used Tasmota devices but now I am getting the hints about using OpenBK devices also in this setup.

Thanks.