openshwprojects/OpenBK7231T_App #895: Adjust HomeAssistant EntityName to confirm to changes in 2023.8.1

iprak, 2023-08-09:
**Describe the bug**
HomeAssistant https://www.home-assistant.io/blog/2023/08/02/release-20238/ has a break change in MQTT Entity Name - _The entity's name property only identifies the data point represented by the entity, and should not include the name of the device or the type of the entity. So for a sensor that represents the power usage of its device, this would be “Power usage”._

There was a PSA announcement about the same - https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

**Firmware:**
- Version 1.17.208
- Device? Any
- Chip/model: Any

**Screenshots**
One will see a notification like this under Settings. The message clarifies that after 2024.2.0 this can result in duplicate entities.
[image]

**Additional context**
This can be addressed by changing
`sprintf(g_hassBuffer, "%s %s", CFG_GetShortDeviceName(), CHANNEL_GetLabel(index));` 
to
`sprintf(g_hassBuffer, "%s", CHANNEL_GetLabel(index));`

Here is the MQTT payload for a fixed item which just has '0' as the entityName.
[image]

We can also return `None` as the name if the entity is the main device for LIGHT_PWMCW case where the entityName is the same as deviceName.

openshwprojects, 2023-08-10:
Thanks, I saw that mentioned somewhere and was not sure how to fix it. So that rename sprintf in HA discovery is all we need?

iprak, 2023-08-10:
Sort of.

If name was not assigned then HomeAssistant will auto assign it. For example, for Relay, the default name is "MQTT Switch". The default name for PWM styled entity is [MQTT LightEntity](https://github.com/home-assistant/core/blob/e9f9c7799a5b81be8ce95fcb63bdb969a5bdc3be/homeassistant/components/mqtt/light/schema_basic.py#L135). Simple Light/LED are just switch at HomeAssistant end.

I previously thought that assigning "None" to name will be sufficient for PWM case which currently uses the device shortName as the name. But that does not work, it ends up using the literal "None" as the name. None only works from within custom entities. 

I am thinking of not emitting the name and let the default name get assigned. One can customize the entity in HomeAssistant and give it a more suitable name.

Alternatively, we can use something like "Light" but one might still have to customize it.

This might be a slightly breaking change. Note that the entity's unique id is not changing so the entity will still continue to be referenced as before.  

I can submit a merge soon and you can try it out.

jenshenk, 2023-08-11:
Thanks for looking into this, yet I'm still confused by this and wonder if the above proposed solution would really fix the error message / warning in  home assistant. What confuses me the most is that the error message does not seem to be true, at least how I understand it. For example: 

Here's a smart socket I have and its HA discovery mqtt message:
```
mqtt:
  switch:
  - unique_id: "LSC_01_relay_1"
    name: "LSC_01 1"
    state_topic: "openbk/lsc_01/1/get"
    command_topic: "openbk/lsc_01/1/set"
    qos: 1
    payload_on: 1
    payload_off: 0
    retain: true
    availability:
      - topic: "openbk/lsc_01/connected"
```

So in this example there's a name (instead of a 0) and there is no "switch" in the name. But, because it is published as switch type, the entity is automatically being named switch.lsc_01_1 as per HA standards - in recent HA versions light, switch, sensor is actually greyed out and cannot be edited. 
This made me think that maybe the error message is a false positive? Maybe HA misidentifies how the name is created? Maybe actually the mqtt integration somehow is responsible? Still some time to fix it (the warning said by 2024.2 it will be a real problem) but unless I'm overlooking something it seems like there's not a lot that could be done from the openbeken side.

jenshenk, 2023-08-11:
Oh wait ... could it be that whatever creates the error message is not correctly looking for light, switch, sensor at the beginning of the device name but instead if it's part of the name at all?

Would be interesting to see if the message goes away if the published entities do not have e.g.  _light_01 at the end

jenshenk, 2023-08-13:
Felt like adding: I Just read through the whole PSA post and the comments (link was shared above) https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099 and the most comforting info I read was that the worst thing that will happen is that as openbeken is publishing names at this moment, the initial name of a device in HA called e.g. "Ceiling Light" where the relay would be published as "Ceiling Light 01" would become "Ceiling Light Ceiling Light 01". So I'd say next to an automated option, a fix and also a nice addition in general would be to have the "Friendly Name" fields like under "Configure Other" in Tasmota.

openshwprojects, 2023-08-13:
@iprak feel free to submit a merge and we'll see how it goes. @jenshenk , it's possible to download binary from pull request without merging it so you will be able to test it early as well

jenshenk, 2023-08-13:
cool, thanks!

openshwprojects, 2023-08-21:
@iprak do you know if this error is related?
https://www.elektroda.com/rtvforum/viewtopic.php?p=20701723#20701723

iprak, 2023-08-21:
Based on the MQTT payload, it seems that index is incorrect or the if block checks are incorrect in `hass_init_power_sensor_device_info`. If the execution does not enter any of the if blocks then the `state_topic (stat_t)` would be missing. Note that state_class is also missing which is also defined in those blocks.

closed 2023-08-23 by iprak