openshwprojects/OpenBK7231T_App #897: Home Assistant MQTT persistent config not updated when IP address changes

rspier, 2023-08-13:
**Describe the bug**
When the IP address of a device changes (i.e. DHCP lease expires while device is off for a week or three), the `homeassistant/light/OpenBK7231N_51F2XXXXX_light/config` message is not updated, so the VISIT links inside HA won't work.

**Firmware:**
- Version 1.17.213

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a device.  Configure it to work in HA.  The "VISIT" button will work.
2. Change the IP address.  (Either by hand, or through dhcp expiration.)
3. The device will still function as it can receive/send MQTT messages, but the "VISIT" button won't, because the persistent MQTT message will still have the old value.

**Additional context**
Workaround: Go to each device and click "Start Home Assistant Discovery" to get a new `/config` message sent.

openshwprojects, 2023-08-14:
Hello, do you have OBK_FLAG_AUTOMAIC_HASS_DISCOVERY flag set? Flag 27?

As far as I can tell, the flag 27 will check for IP change and will redo HASS discovery in that case. So basically, you need to have flag 27 set.

rspier, 2023-08-18:
You're right.  I didn't find that in the docs.

Maybe Flag 27 should be on by default?  (Happy to send a PR if you think it would be useful, it looks straightforward.)