openshwprojects/OpenBK7231T_App #903: mqtt broker running in openbeken

analystcmyk, 2023-08-16:
**Situation :** 

I have 2 openbeken devices. 

I would like to publish an MQTT message from device 1, to be picked up by device 2.  

For that I need an MQTT broker. 

**Question** : 

Would it be possible to run a minimal broker on one of the two?  Or would you just use the http api ?

I was looking at https://github.com/martin-ger/uMQTTBroker/tree/master 

**Examples :**  

- We have one master switch, and 4 slave switches. Switching on the master should switch on all slaves.
- A power meter publishes the total power. a slave device should switch off when the total power goes above a max value.

openshwprojects, 2023-08-16:
Interesting idea, but why not try the SendGet approach?

analystcmyk, 2023-08-16:
well, yes, but : 
- sendget would mean all slaves continuously polling and/or vice versa
- sendget would mean master must know all slaves (?), would that mean setting dns mac address reservation ?
- with mdns you would only configure the masterOBK.local name for the "master" device, then all slaves would use that as broker.

Would it even be possible regarding memory ? 

Maybe out of scope but fun to think about how you could do load balancing with just 2 or 3 OBK devices, without any additional equipment required.

Which reminds me to bump #711

analystcmyk, 2023-08-17:
Hmm mDNS vs SSDP : mdns would allow to use the .local as hostname for the MQTT broker, but SSDP doesnt. 
I think mDNS is better supported if you have Mac clients (bonjour) right, compared to SSDP 

I will study device groups first, thank you for the hint.