openshwprojects/OpenBK7231T_App #904: Sticky WiFi client on BK7231N

3xtant, 2023-08-18:
I have multiple APs in my network with the same SSID, so that clients can roam between them with even coverage. I have experienced the same issue with Tasmota where new devices connect to the first AP where they are configured and then refuse to connect to other APs, even when they are much closer and have stronger signals.

In the case of Tasmota, one workaround is to add the same SSID as a secondary WiFi network. Devices try to connect to the first AP, but then switch to a stronger one. I tried the same thing here without success. The device simply refuses to connect to other APs. Reflashing (OTA) the WiFi configuration section did not solve the issue either.

The same issue may exist on BK7231T, I just have not had a chance to test it yet.

openshwprojects, 2023-08-18:
Hello, so you're saying that second SSID is not working?
Well, I apologize, it seems it's just not implemented yet.
I have to find time to finish it.
The field exists in the config, but it's not used.

Can you help with testing, once I submit the missing part of the code?

3xtant, 2023-08-19:
Sure, I'll be happy to!