openshwprojects/OpenBK7231T_App #905: Device Group with Tuya Dimmer EDM-01AA-EU - Channel 2 (Dimmer) Not Working with Tasmota Devices

Hollako, 2023-08-18:
Hello,

First of all, I want to thank you for the great effort you in this project.

I want to highlight an issue I am facing with the device group of the Tuya Dimmer EDM-01AA-EU where I activated the device group in Tasmota by enabling the so85 and so88. Only the On/Off Channel is working however the Dimmer channel is not working in the device group. here is some information about the device am working on.

- Version: 1.17.210
- Device: Tuya Dimmer EDM-01AA-EU
- Chip/model: BK7231T

On the other hand, I tried to find a workaround by using the "addEventHandler" commands and then publishing a specific MQTT topic to be able to control the Tasmota Dimmer by using the below command:

"addEventHandler OnChannelChange 2 Publish LED_Strip/cmnd/Dimmer2 $led_dimmer"

However, I found out that OpenBK firmware doesn't publish the topic as it's listed in the command. it's all time adding a Prefix (Device_Name) and Suffix (get) to this topic and this is mentioned in the documents as well. My question here is it possible to remove this Prefix and Suffix when publishing a custom MQTT topic?

Thank in advance

openshwprojects, 2023-08-18:
Hello, so you basically want a command like "PublishRaw", without those prefixes? No problem, I can add that tomorrow.

Hollako, 2023-08-19:
Hello,

Thanks for the quick reply.

Correct without prefixes and suffixes, only the topic that I included in the command. in my opinion, it gets the firmware more flexible and removes the limitation of sending the device name with each published MQTT topic.

Best Regards,

openshwprojects, 2023-08-19:
ok i added third arg:
[image]
if set to 1, it results in given topic used directly:
[image]
can you check lates tbuild?

Hollako, 2023-08-19:
Sure no problem, 

Just can you guide me where I can find latest build file ?

openshwprojects, 2023-08-19:
sorry, automatic build didnt trigger, try now, check releases tab

Hollako, 2023-08-19:
Hello,

Tested and all good. many thanks !!

But I am facing another issue not related to this one, am not sure if I should open another issue or continue in here. The issue is that when I restart the dimmer it's connecting again to Wifi, so I keep powering Off the dimmer many times and then On to get it online again. Please let me know where I should start to help in troubleshooting this issue.

Regards,

openshwprojects, 2023-08-19:
Better open another issue for clarity and please explain in detail whats wrong, because current explanation is not clear to me.