openshwprojects/OpenBK7231T_App #907: Device does nor remember previous stare (Tuya water valve)

xszuflax, 2023-08-20:
**Describe the bug**

Hello the device I'm using dies not remember last states: I set -1 value in /cfg_startup Channel 0: but every time power is lost it goes to valve closed state.

I believe this was working around  1.17.108

**Firmware:**
- Version  1.17.224
- Device Tuya smart water valve 12V 1A (Built it template form the App) 
- Chip/model: BK7231T (WB3S)
- Device config: 

  "pins": 
    "1": "WifiLED_n;0",
    "6": "Rel;0",
    "24": "Btn;0"

**To Reproduce**
Steps to reproduce the behaviour:
0. Set the device to on  (Valve open)
1.  Disconnect the power cable
2. Connect the power cable
3. Device goes to the state closed valve instead of keeping it open

openshwprojects, 2023-08-20:
Hello,  are you sure that channel forgets its value? Maybe the problem is caused by the short time when relay is off during reboot?

Maybe,for the sake of just a test, please remove Rel role from P6 and change P1 role to Rel.

Then make sure that channel 0 has -1 start value.

Then check if device remembers the LED state between reboots.

xszuflax, 2023-08-20:
OK,

now when u put it that way. I do not care about LED. I do care about relay which is on 6. How can I set it to remember the stare? 

My config screen looks like this:

[image]

xszuflax, 2023-08-20:
I set it up like this: 

[image]

And the led Goes off after the power reconnect. The Arm also goes to shut down valve position.

openshwprojects, 2023-08-20:
The problem is, it works for me. I tested right now. Something must be different in our configurations.
[image]

Do you have MQTT  connected? Maybe try to disable MQTT?

xszuflax, 2023-08-20:
I do have MQTT enabled and following startup flags:

Flag 2 - [MQTT] Broadcast self state every N (def: 60) seconds (delay configurable by 'mqtt_broadcastInterval' and 'mqtt_broadcastItemsPerSec' commands)
Flag 10 - [MQTT] Broadcast self state on MQTT connect
Flag 40 - [MQTT] Use kWh unit for energy consumption (total, last hour, today) instead of Wh

I will try with MQTT but I doubt it will change anything. The default state is open for this valve so if MQTT was to update it would keep it open. And WiFi is not there yet when the valve is closing after a power reboot. The LED blinks.

It was doing this on a Tuya firmware as well. 

I remember it was working with OpenSHW. Once rebooted it was triggering for a split sec but the valve was not closing. IMO something changed between 107 and the newest FW.

openshwprojects, 2023-08-20:
so maybe please backup your config (json and flash config partition) and try to downgrade to check?

or maybe do you have some scripts in autoexec.bat?

xszuflax, 2023-08-20:
I do not have any customisations in autoexec.bat. To be honest with you I'm fairly new to the OpenSHW. O bought this Tuya valve hoping it will be ESP8266 based so I could use [HAA ](https://github.com/RavenSystem/esp-homekit-devices) . I do use HomeKit by default. 

I've only set a template, configured MQTT and enabled these 3 startup flags. Nothing more. Maybe I just do a factory reset and start re-configuring this unit once again. One option at a time.

Thank you for your help.

xszuflax, 2023-08-21:
OK so this is wired. I put my config back to the device startup flags and MQTT and it is still working. 
Maybe one of the updates broke it. Not sure.

closed 2023-08-21 by xszuflax