openshwprojects/OpenBK7231T_App #908: Tuya Dimmer EDM-01AA-EU Not Connecting to Wifi after Reboot

Hollako, 2023-08-20:
Hello, hope all is good.

I am facing an issue with the Tuya Dimmer EDM-01AA-EU where it's not connecting to Wifi after reboot and I keep turning it Off and On many times before the Dimmer is connected again to Wifi. Once connected, the dimmer will work properly without any issues.

Version: 1.17.222
Device: Tuya Dimmer EDM-01AA-EU
Chip/model: BK7231T

And here is below the autoexec.bat file script:

startDriver TuyaMCU
setChannelType 1 toggle
setChannelType 2 dimmer
tuyaMcu_setBaudRate 115200
tuyaMcu_setDimmerRange 1 1000
linkTuyaMCUOutputToChannel 1 bool 1
linkTuyaMCUOutputToChannel 2 val 2
addEventHandler OnChannelChange 1 Publish LED_Strip/cmnd/Power2 $CH1 1
addEventHandler OnChannelChange 2 Publish LED_Strip/cmnd/Dimmer2 $CH2 1

Kindly advise where I should start to troubleshoot this issue.

Regards,

openshwprojects, 2023-08-20:
Hello, does it happen without custom event handlers?

Hollako, 2023-08-22:
Hello,

I tried it without the eventhandler it's same

openshwprojects, 2023-08-22:
It is normal. Boot incomplete is a boot that didnt last longer than N seconds (default: 30). Every boot is incomplete before timer reaches that time.

Can you empty autoexec.bat and try again?

openshwprojects, 2023-08-22:
No idea what's wrong, but it's always a good practice to put PowerSave 1 in the script