openshwprojects/OpenBK7231T_App #909: BL602: RGBW controller gets stuck after around 5 minutes

biggoof1, 2023-08-21:
**Describe the bug**
Device with BL602 worked about 3 months and controls a LED RGBW stripe.
Out of a sudden the device stops working after around 5 minutes with just not reacting to any command (MQTT, Webpage, ..) any more.
Device gets controlled via MQTT commands.
Flags used are: 0,2,16,18
Device updated to the latest firmware.
When the device gets stuck the LED stripe gets activated in a light bright white color without any submitted command.
Removing power and restoring it reactivates the device for around 5 minutes.
Log attached from the web app.

**Firmware:**
- Version 1.17.225 
- Device? 
- Chip/model: BL602
- Device config Flags 0,2,16,18 / Flags: 327685

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on and wait
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2023-08-22:
Can you please try without mqtt? remove mqtt credentials from settigs

biggoof1, 2023-08-25:
Hey.. 
first of all I was not able to delete the MQTT settings.. Although removing them they reappeared after the page loaded. I was able to change user and pwd instead to remove the connection. Can you please cross check?

The overall behaviour was the same though.
It did live for 6 minutes or so and then got stuck with a light glowing LED stripe.

[image]

biggoof1, 2023-08-26:
I did a factory reset of the device and set it up without the MQTT information but adding the PWM info and the flags.
It is running at the moment without a connected LED stripe and is up for 15 minutes right now..
Let´s see what happens..