openshwprojects/OpenBK7231T_App #910: Voltage spikes with BL0937

giedriuslt, 2023-08-23:
**Describe the bug**
Voltage spikes are reported frequently in 30s intervals. This is happening via mqtt with LSPA9 like plug. Plug are with zero load, although same happens with 100W load too.
The spikes in example  occur in 21:31:56, 21:32:26, 21:32:56. They always occur at the time XX:26 or XX:56 and is always one reporting interval long (6s) but not every minute. 
They must be from device, because other plug in the same socket reports same spikes just on different times. Pzem-004t reports no spikes at all. 
After restart this second changes.
I wonder what could be causing this,  I suspect some periodic job makes BK to miss interrupts?
- Device config?
PowerSave 1, mqtt enabled

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

giedriuslt, 2023-08-26:
Seems like PowerSave is causing this. Here is comparison with and without:
[image]
Violet line is plug with power save, dark blue is without.
Without powersave voltage matches Pzem-004 very well
[image]

openshwprojects, 2023-08-26:
You may have a point. I don't think we've tested PowerSave 1 in depth with BL0937. Does it happen on every device with PS on?

giedriuslt, 2023-08-26:
I just tested it with two more plugs I have,  RCM004 and RMC021 markings. They do have the same behaviour with spikes on the same interval. 
However, I also noticed that with Powersave voltage is reported lower than pzem-004 while after disabling Powersave it gets generally higher. That suggest some interrupts are always skipped with Powersave not only during those spike periods.
See image, after 20:00 Powersave was disabled on both plugs. One of them is not calibrated, but that it not relevant here.
[image]

giedriuslt, 2023-08-27:
Did some testing with mqtt disabled. It's not documented how to disable mqtt and not possible from the UI, should be fixed.
Had to run this command
`MqttHost ""` 
Reading every 5 seconds got the same spikes at minute interval.
These are times with spikes
Time | Voltage
-- | --
20:45:45 | 213,4
20:48:45 | 215
20:49:15 | 210,3
20:49:46 | 217,5
21:04:14 | 214,5
21:04:44 | 214,7
21:05:14 | 208,8
21:13:16 | 204,8
21:13:46 | 213,8
21:14:16 | 208,9
21:14:46 | 212,6
21:15:16 | 210,7
21:15:46 | 210
21:16:16 | 212,9
21:16:46 | 207,7
21:17:16 | 210,6
21:18:01 | 214
21:36:14 | 215,2
21:43:16 | 199,4
21:43:46 | 207,2
21:44:16 | 207,3
21:44:46 | 208,4
21:45:16 | 211,5
21:45:46 | 211,8
21:46:16 | 211,8
21:47:16 | 210,7
22:01:14 | 212,7
22:02:14 | 215,4
22:02:45 | 212,8
22:03:15 | 212,1
22:03:45 | 213,7
22:04:15 | 218,7
22:05:15 | 216,4
22:05:45 | 211,1
22:13:46 | 207
22:14:16 | 212,2
22:14:46 | 212,7
22:15:16 | 212,6
22:15:46 | 214,2
22:16:16 | 214,5
22:16:46 | 217,8
22:17:16 | 212,2
22:17:46 | 215,5
22:18:16 | 211,4

[image]
Reading at 1 second interval seems to work fine, I guess thats too much activity for plug to enter powersave.

openshwprojects, 2023-08-27:
How do you do the reading?

Thank you for the feedback. I've:
- added MQTT info to the page and to FAQ
- added PowerSave BL0937 issue to FAQ
- fixed HTML page so it's possible to submit empty MQTT server to clear it

giedriuslt, 2023-08-27:
Thank you for help.
I used simple python script:
```
import requests
import json
import re
import datetime
import time
import os.path
from datetime import date, timedelta
from lxml.html import document_fromstring

def main():
    file = open('pluglog.csv', 'a')
    while True:
        dt = datetime.datetime.utcnow().isoformat();
        #print(dt)
        resp = requests.get(
            'http://openbk7231n_5541802a/index?state=1',  
            )
        #print(resp.text)
        resp.raise_for_status()
        a = re.search(r'Voltage.*?[>](\d+[.]?\d+?)[<][/]td', resp.text)
        print(dt,";", a.group(1))
        if a:
            file.write(dt+";"+ a.group(1)+'\n')
            file.flush()
        time.sleep(5-resp.elapsed.total_seconds())

if __name__ == "__main__":
    main()
```