openshwprojects/OpenBK7231T_App #911: CRC error on flashing BK7231N

Lexcis, 2023-08-25:
When flashing a BK7231N from Tuya Universal IR remote (model S06 from aliexpress, PCB is named "CB3S") I get a CRC error at the end of the flashing process. The firmware does not seem to start after repowering. I tried both programs - BK7231Writer and BK Flasher.
The result is always something like "Send by BK 0xAAE42DD2, our CRC 0x175C7EE6" or "Send by BK 0xE8A533DE, our CRC 0x175C7EE6"

I recently successfully flashed a BK7231T from a similar device (also S06 from aliexpress with PCB named "WB3S").

**Firmware:**
- Version OpenBK7231N_QIO_1.17.226.bin
- Universal IR remote S06 (PCB version CB3S)
- Chip/model: BK7231N

[image]
[image]
[image]

openshwprojects, 2023-08-25:
Can you attach 2MB ddump from this device?

Maybe it's BK7231T with wrong label?

Lexcis, 2023-08-25:
How do I dump it? I tried "Do firmware backup (read) only". Same CRC error. Also tried selecting BK7231T in config.
Maybe it's a fake chip?

Backup name is set to 20230825_BK7231N.
Starting read!
Now is: Freitag, 25. August 2023 11:38:19.
Flasher mode: BK7231N
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... Ok! Reading 0xA3000... Ok! Reading 0xA4000... Ok! Reading 0xA5000... Ok! Reading 0xA6000... Ok! Reading 0xA7000... Ok! Reading 0xA8000... Ok! Reading 0xA9000... Ok! Reading 0xAA000... Ok! Reading 0xAB000... Ok! Reading 0xAC000... Ok! Reading 0xAD000... Ok! Reading 0xAE000... Ok! Reading 0xAF000... Ok! Reading 0xB0000... Ok! Reading 0xB1000... Ok! Reading 0xB2000... Ok! Reading 0xB3000... Ok! Reading 0xB4000... Ok! Reading 0xB5000... Ok! Reading 0xB6000... Ok! Reading 0xB7000... Ok! Reading 0xB8000... Ok! Reading 0xB9000... Ok! Reading 0xBA000... Ok! Reading 0xBB000... Ok! Reading 0xBC000... Ok! Reading 0xBD000... Ok! Reading 0xBE000... Ok! Reading 0xBF000... Ok! Reading 0xC0000... Ok! Reading 0xC1000... Ok! Reading 0xC2000... Ok! Reading 0xC3000... Ok! Reading 0xC4000... Ok! Reading [... cut ...] ng 0x1F5000... Ok! Reading 0x1F6000... Ok! Reading 0x1F7000... Ok! Reading 0x1F8000... Ok! Reading 0x1F9000... Ok! Reading 0x1FA000... Ok! Reading 0x1FB000... Ok! Reading 0x1FC000... Ok! Reading 0x1FD000... Ok! Reading 0x1FE000... Ok! Reading 0x1FF000... Ok! 
Basic read operation finished, but now it's time to verify...
Starting CRC check for 512 sectors, starting at offset 00
CRC mismatch!
Send by BK 0x1B42421F, our CRC 0x84AB7CBB
Maybe you have wrong chip type set?
There was no result to save.

Lexcis, 2023-08-25:
Dumping with BK Writer was successful. Resulting file is 1MB. So it's probably not a suitable chip version for OpenBK?!

openshwprojects, 2023-08-25:
In BKWriter, you can set read length to 2 megabytes.

Lexcis, 2023-08-25:
Tried to write it with BK Writer again. There it says "deprotecting flash failed".

For dumping I set "Operate Length" to 0x200000. Hope that is correct for 2MB.

[image]
[image]
[bk7231s_dump-2023- 8-25-13-43-28.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/12438773/bk7231s_dump-2023-.8-25-13-43-28.zip)

openshwprojects, 2023-08-25:
Try this method for N:
https://www.youtube.com/watch?v=PKkiqDNFIx8&ab_channel=Elektrodacom

Lexcis, 2023-08-25:
Ok, I'll try it next week. Thank you.

Lexcis, 2023-08-29:
Doesn't work either. Same error.
Problem is solved anyway. I have some ESP8266 laying around and will replace the CB3S module. Additionally I ordered a few more universal remotes hoping they will contain ESP or BK7231T. For 4 USD each it is not worth trying that long.
Thank you.

[image]

q1111111, 2023-11-15:
I get a similar error on the same chip, what could be the solutions