openshwprojects/OpenBK7231T_App #921: Is RfSend command supported?

anistor, 2023-09-12:
I'm looking for functionality similar to or identical (compatible) with RfSend from Tasmota (see https://tasmota.github.io/docs/Commands/#rf-transceiver)

I'm mostly interested in being able to send short RF messages to emulate an RF remote. Receiving is not important to me atm.

I've already noticed there is support for IRsend but could not find anything for RFSend so was wondering.

Thanks!

openshwprojects, 2023-09-12:
I don\t have any RF devices yet, so not, but it's planned. It should be easy to get that working, if needed. Would you like to help?

anistor, 2023-09-13:
I do not have any rf device either. I have a universal tuya IR remote S06 like this one: https://www.elektroda.com/rtvforum/topic3961676.html

I opened it and found there's enough room in the case so, I'm planning to modify it by adding a 433Mhz rf tx module like this one: https://components101.com/modules/433-mhz-rf-transmitter-module

Unfortunatelly the space is not enough to also add a receiver module, but I personally would not need one because I was able to receive and decode my remotes I'm planning to use with the help of an rtl-sdr dongle and the rtl_433 tool, so I know the codes. So all I need is support for RfSend command via REST and Mqtt.

I think I could try to take the existing IR driver and rip out the IR related parts and then try to stuff in the rc-switch arduino library to add support for sending RF codes. Or is there a better way to start a new driver from scratch?

But I'm really low on available time, so not sure when I can complete this grand plan :) I'll surely make a PR when I have it.

maalouf, 2023-10-25:
Hello, i am also interested in porting RC-switch to the openBk app project. 
can you point me however to some useful information that can help. thank you